dkms-project/dkms #248: Where can I find a dkms-template?

trialley, 2022-09-20:
Hi! I am currently working on a dkms kernel module. It seems that there is no dkms module template, where can I get one?

scaronni, 2022-09-21:
Just the `dkms.conf` file? Or the entire packaging step? If yes, for which distribution?

evelikov, 2022-09-27:
We've tried having templates in-tree but they were badly outdated, thus got removed. I would suggest picking an up-to date dkms package from your distribution of choice and using that as a template. 

From a quick search here are two [v4l2loopback](https://salsa.debian.org/debian/v4l2loopback/-/tree/master/debian) and [r8168](https://salsa.debian.org/debian/r8168/-/tree/master/debian). Note I haven't tried either of them - Debian isn't my cup of tea.

closed 2022-09-27 by evelikov