dkms-project/dkms #249: Moving repository

scaronni, 2022-09-27:
Now that admin access has been restored and this is not really a Dell project, shall we move it to a proper organization on its own or some other already existing organization?

What do you think?

evelikov, 2022-09-27:
Sounds reasonable to make it a separate org. Here are some benefits that I can see off the top of my head:
 - visibility of project members - github shows org people, but not project people
 - separate designated repos - dkms itself, tooling (say releasing script, other), Debian templates, Redhat templates ... other 
   - mentioned templates, since it was brought up a couple of times
 - distinct (sub)set of people per repo (as needed of course) - for example, add a bunch of Debian or RH/Fedora devs for their templates, so they can manage as they see fit

scaronni, 2022-09-27:
Moved to discussions here, I think it's a better place: https://github.com/dell/dkms/discussions/251

Can you please add your thoughts there? Thanks!

closed 2022-09-27 by scaronni