dkms-project/dkms #256: ArchLinux module signing

Ouack23, 2022-10-09:
Hello, 

I maintain my own linux custom kernel which includes out of tree kernel modules signing.
I used the `sign-file` variable in `/etc/dkms/framework.conf` for quite some time before the whole signing mechanism changed.
I painfully adapted my `framework.conf` to use the new dkms `mok` signing thing, and it worked for some time.
It seems it is broken again, so I'm posting this issue here.

Here is my current `framework.conf` :
```
...

# Location of the sign-file kernel binary (default: depends on distribution):
# sign_file="/path/to/sign-file"
#sign_file="/usr/bin/sign_manual.sh"
#sign_file="/usr/lib/modules/$kernelver/build/scripts/sign-file"

# Location of the key and certificate used for Secure boot (default: /var/lib/dkms):
mok_signing_key='/usr/lib/modules/$kernelver/build/certs-local/signing_key.pem'
mok_certificate='/usr/lib/modules/$kernelver/build/certs-local/signing_crt.crt'

# Automatically modprobe the built modules upon succesful installation:
# modprobe_on_install="true"
```

The `$kernelver` variable in the `mok` lines seems not to be interpreted as it was in last dkms version.

I have trouble signing my modules, here is an error log with the nvidia module (same diagnostic for all dkms modules) :
```
==> dkms install --no-depmod nvidia/515.76 -k 5.19.14-hardened1-1-hardened-with-keys
At main.c:167:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: /usr/lib/modules/$kernelver/build/certs-local/signing_key.pem: No such file or directory
At main.c:167:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: /usr/lib/modules/$kernelver/build/certs-local/signing_key.pem: No such file or directory
At main.c:167:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: /usr/lib/modules/$kernelver/build/certs-local/signing_key.pem: No such file or directory
At main.c:167:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: /usr/lib/modules/$kernelver/build/certs-local/signing_key.pem: No such file or directory
At main.c:167:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: /usr/lib/modules/$kernelver/build/certs-local/signing_key.pem: No such file or directory
```

xuzhen, 2022-10-11:
I committed a patch for this. Can you try it https://github.com/xuzhen/dkms/blob/patch-1-1/dkms.in

evelikov, 2022-10-13:
@xuzhen we should really be looking at reducing `eval` usage not adding more IMHO.

@Ouack23 are you seeing this with the official Arch kernels and something simpler - say bbswitch-dkms? If you are then, I can take a look into it. Otherwise it sounds like an issue with your custom kernel package.

xuzhen, 2022-10-14:
@evelikov This problem has nothing to do with custom kernels.

The root of the problem is that the /etc/dkms/framework.conf file is [executed at the beginning of the program](https://github.com/dell/dkms/blob/6a35e8af81368627518123ccf442fbc009971dea/dkms.in#L2217), when variables such as kernelver have not been initialized. So you can't just put ```sign_file="/path/with/$kernelver/"``` in /etc/dkms/framework.conf.

From this line, putting single quotes around the value of sign_file/mok_signing_key/mok_certificate is intentionally supported IMHO.  https://github.com/dell/dkms/blob/6a35e8af81368627518123ccf442fbc009971dea/dkms.in#L1030 

But ```sign_file='/path/with/$kernelver/to/binary'``` without variable expansion will fail the existence check before the ```eval```https://github.com/dell/dkms/blob/6a35e8af81368627518123ccf442fbc009971dea/dkms.in#L879-L882

```mok_signing_key='/path/with/$kernelver/to/key'``` and ```mok_certificate='/path/with/$kernelver/to/cert'``` will also be passed to ```openssl``` without variable expansion
 https://github.com/dell/dkms/blob/6a35e8af81368627518123ccf442fbc009971dea/dkms.in#L919-L924

My patch is to use eval to expand the variable before checking the existence of the files. We can also execute /etc/dkms/framework.conf after variables such as kernelver are set.

Or we just prohibit this usage. The users like @Ouack23 can set ```sign_file="/usr/bin/sign.sh"``` in /etc/dkms/framework.conf, and the sign.sh is something like
```
#!/bin/sh
"/usr/lib/modules/$kernelver/build/scripts/sign-file" "$1" "/usr/lib/modules/$kernelver/build/certs-local/signing_key.pem" "/usr/lib/modules/$kernelver/build/certs-local/signing_crt.crt" "$4"
```

Ouack23, 2022-10-16:
Hi, thanks @xuzhen for your replies. You perfectly understood what is happening. Variables are not replaced by their values, and therefore the script cannot find my cryptographic key / cert.

I'm sorry not to have tested your patch yet, I haven't had the time to try it for now. But from the looks of it, it seems to be good :pray:. I'll report here when I have some time to test it.

I don't get why it would be a good idea to prohibit variable usage in `framework.conf` entries, as they can be directly linked to the kernel you want to sign the module from.
IMO `framework.conf` parsing should be happening after variables are set, but I guess I don't see the whole picture here and it may be way more complicated than that ...

For your `sign_file` workaround, it is nice, but hasn't the `sign_file` entry disappeared from `framework.conf` since august updates ?

xuzhen, 2022-10-17:
> For your `sign_file` workaround, it is nice, but hasn't the `sign_file` entry disappeared from `framework.conf` since august updates ?
sign_file was introduced in https://github.com/dell/dkms/commit/3323033dfdc5f58afc303736f6b580bf62442ef7#diff-c40e933801c10a3c3bd64c875d2f2b7d8fd8a8a0e3667e3d1e64aa78b56a7535 and replaced sign_tool.

evelikov, 2022-10-18:
Sorry my bad - was reading the report, while before the morning coffee. Yes it's unrelated to custom kernels and dkms modules.

Thinking out loud - seems like the issue should have been caught in our CI tests. Are we missing test coverage for this?
I am still inclined that one should reduce the use of eval not add more... But I'm not coming with any clever ideas, so don't read too much on that.

xuzhen, 2022-10-18:
I opened the pull request #260 for this issue.
The /etc/dkms/framework.conf will be executed again after variables such as kernelver have been initialized. No longer need to use single quotes, ```mok_signing_key=/path/with/$kernelver/to/key``` is enough.

xuzhen, 2022-10-27:
The pull request #265 was merged. @Ouack23 you can use double quotation marks for your use case now.
```
mok_signing_key="/usr/lib/modules/$kernelver/build/certs-local/signing_key.pem"
mok_certificate="/usr/lib/modules/$kernelver/build/certs-local/signing_crt.crt"
```

closed 2022-10-27 by xuzhen