dkms-project/dkms #266: sha512 signed modules not works

misterhsp, 2022-10-26:
Modules signed with the dkms-3.0.8 are rejected and I end up at the prompt of initramfs. In my case zfs-modul. The modules are signed but not accepted. Neither with my own keys nor the ones created by dkms. I am using Debian SID

The reason for this can be found in the following line.

eval '"$sign_file" sha512 "$mok_signing_key" "$mok_certificate" "$built_module"'

Here I get hash_algo sha512 signed which doesn't work on my box and my
laptop. If I change this to sha256 secure-boot works as desired. Why
this is so I can not say, this is beyond my knowledge. Also if this is a
bug or not I can't say.

Sorry for my crap english :-)

...

evelikov, 2022-10-26:
What is the error message that you see as result of the above line? What distribution/version are you running?

misterhsp, 2022-10-26:
I don't see any error message, the modules are signed. But they are not
accepted. I use Debian SID, but it is currently only 3.0.6, but the same
thing happens. The Debian modules all have sig_hashalgo = sha256

# modinfo vfat
filename:       /lib/modules/6.0.0-2-amd64/kernel/fs/fat/vfat.ko
author:         Gordon Chaffee
description:    VFAT filesystem support
license:        GPL
alias:          fs-vfat
depends:        fat
retpoline:      Y
intree:         Y
name:           vfat
vermagic:       6.0.0-2-amd64 SMP preempt mod_unload modversions
sig_id:         PKCS#7
signer:         Debian Secure Boot CA
sig_key: 32:A0:28:7F:84:1A:03:6F:A3:93:C1:E0:65:C4:3A:E6:B2:42:26:43
sig_hashalgo:   sha256
signature: 43:87:99:64:C0:DE:8F:02:F0:A2:17:F6:5C:AB:45:64:0F:C6:66:E6:
E6:31:BB:95:86:BE:B7:E1:3C:2D:8B:FA:40:E8:12:66:A7:6D:3A:2E:
81:A4:CE:56:F9:BF:B6:E5:5D:91:F1:AA:05:13:0B:B9:FB:75:25:89:
CD:20:C3:61:37:5C:48:92:F1:16:E9:A8:B8:F0:A9:A0:B8:A4:93:01:
C1:68:66:66:9E:3F:33:15:10:51:0D:29:A5:36:B9:46:74:D6:8F:B2:
B8:DA:F9:8E:E9:C3:85:AC:F4:15:ED:45:5A:CA:8E:AF:41:AD:1E:B9:
45:42:CF:56:69:CF:E0:3F:A7:72:79:63:E3:61:A0:AD:E4:5A:AF:37:
CA:17:12:FB:83:FA:E1:C8:1E:00:85:E8:25:A3:5F:CD:D5:BF:4A:63:
23:3D:BB:C3:D0:C2:24:F0:38:17:9E:A9:98:1B:49:78:E6:8E:F0:07:
DA:86:2A:DE:9A:23:DD:8F:CA:89:F8:EE:4E:18:B2:04:43:5D:30:9A:
8F:D0:C4:6F:9E:30:74:2D:A1:7A:CF:5B:43:38:D6:3D:C6:0F:35:66:
F2:FA:AE:75:09:15:E3:4A:91:92:39:92:58:DA:04:22:0C:59:9F:D8:
         9D:15:06:A3:E8:2A:13:DE:FB:F9:D1:7A:2D:40:94:5D

xuzhen, 2022-10-26:
There is a kconfig option for the hash algorithm of module signature. ```grep CONFIG_MODULE_SIG_HASH /boot/config-$(uname -r)```

dkms should follow that option, or at least add a hash configuration in framework.conf.

misterhsp, 2022-10-26:
Following the kconfig option would probably be better than a
configuration in framework.conf.

misterhsp, 2022-10-27:
> Pulled https://packages.debian.org/sid/kernel/linux-config-6.0 and
> confirmed that |CONFIG_MODULE_SIG_HASH| is "sha256".
>
> @xuzhen <https://github.com/xuzhen> bear in mind the config is not
> guaranteed to live in |/boot/config-XXX|, is in
> |/usr/lib/modules/uanme_a/build/.config| as mentioned in #269

grep CONFIG_MODULE_SIG_HASH /boot/config-6.0.0-2-amd64
CONFIG_MODULE_SIG_HASH="sha256"

misterhsp, 2022-11-04:
I have of course immediately tried with the new commit and must say it is still signed with sha512 here with me. secure-boot not possible.

modinfo vfat
filename:       /lib/modules/6.0.0-2-amd64/kernel/fs/fat/vfat.ko
author:         Gordon Chaffee
description:    VFAT filesystem support
license:        GPL
alias:          fs-vfat
depends:        fat
retpoline:      Y
intree:         Y
name:           vfat
vermagic:       6.0.0-2-amd64 SMP preempt mod_unload modversions
sig_id:         PKCS#7
signer:         Debian Secure Boot CA
sig_key: 32:A0:28:7F:84:1A:03:6F:A3:93:C1:E0:65:C4:3A:E6:B2:42:26:43
sig_hashalgo:   sha256
signature: 43:87:99:64:C0:DE:8F:02:F0:A2:17:F6:5C:AB:45:64:0F:C6:66:E6:
E6:31:BB:95:86:BE:B7:E1:3C:2D:8B:FA:40:E8:12:66:A7:6D:3A:2E:
81:A4:CE:56:F9:BF:B6:E5:5D:91:F1:AA:05:13:0B:B9:FB:75:25:89:
CD:20:C3:61:37:5C:48:92:F1:16:E9:A8:B8:F0:A9:A0:B8:A4:93:01:
C1:68:66:66:9E:3F:33:15:10:51:0D:29:A5:36:B9:46:74:D6:8F:B2:
B8:DA:F9:8E:E9:C3:85:AC:F4:15:ED:45:5A:CA:8E:AF:41:AD:1E:B9:
45:42:CF:56:69:CF:E0:3F:A7:72:79:63:E3:61:A0:AD:E4:5A:AF:37:
CA:17:12:FB:83:FA:E1:C8:1E:00:85:E8:25:A3:5F:CD:D5:BF:4A:63:
23:3D:BB:C3:D0:C2:24:F0:38:17:9E:A9:98:1B:49:78:E6:8E:F0:07:
DA:86:2A:DE:9A:23:DD:8F:CA:89:F8:EE:4E:18:B2:04:43:5D:30:9A:
8F:D0:C4:6F:9E:30:74:2D:A1:7A:CF:5B:43:38:D6:3D:C6:0F:35:66:
F2:FA:AE:75:09:15:E3:4A:91:92:39:92:58:DA:04:22:0C:59:9F:D8:
         9D:15:06:A3:E8:2A:13:DE:FB:F9:D1:7A:2D:40:94:5D

closed 2022-11-05 by evelikov

misterhsp, 2022-11-05:
Thanks for the quick fix. Now it works.

...

PR #274 (2022-11-04): Follow the target kernel's module hash algorithm
Fixes #266
PR #278 (2022-11-05): Ensure the kconfig file location is set during building
#266 is not fully fixed by #274. This is because the variable `kernel_config` is not set before calling `prepare_signing` if the module was not built before.

Fix #266.