dkms-project/dkms #273: Unable to sign modules on Debian on module install

siddhpant, 2022-10-27:
Unable to sign modules on Debian. It simply won't sign on install. After setting `mok_signing_key` and `mok_certificate`, I get some errors, probably because it doesn't ask for passphrase for the key?

Earlier I used to use `sign_tool` script, which properly worked by fetching passphrase from keyring, [see here](https://gist.github.com/siddhpant/19c07b07d912811f5a4b2893ca706c99#requirements). I had to downgrade dkms because it was not working, but now nvidia drivers require newer dkms, but the newer dkms doesn't sign the modules, so the nvidia driver does not load.

---

Version: Latest master (030a09fc657e66e2e641bbf8ce1c8c1773245355)

Changes to `framework.conf`:

```
mok_signing_key=/var/lib/shim-signed/mok/MOK.priv
mok_certificate=/var/lib/shim-signed/mok/MOK.der
```

Trigger command:

```
sudo apt install nvidia-driver
```

Errors:

```
At main.c:170:
- SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
- SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
- SSL error:1C80009F:Provider routines::unable to get passphrase: ../providers/implementations/encode_decode/decode_epki2pki.c:96
- SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
- SSL error:04800068:PEM routines::bad password read: ../crypto/pem/pem_pkey.c:155
sign-file: /var/lib/shim-signed/mok/MOK.priv: Success
```

System info:

```
Debian Testing

Linux kampyooter 5.19.0-2-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.19.11-1 (2022-09-24) x86_64 GNU/Linux
```

---

<details>

<summary>Full framework.conf file:</summary>

```
# This configuration file modifies the behavior of DKMS (Dynamic Kernel Module
# Support) and is sourced in by DKMS every time it is run.

# Source Tree Location (default: /usr/src):
# source_tree="/usr/src"

# DKMS Tree Location (default: /var/lib/dkms):
# dkms_tree="/var/lib/dkms"

# Install Tree Location (default: /lib/modules):
# install_tree="/lib/modules"

# Temporary folder Location (default: /tmp):
# tmp_location="/tmp"

# Verbosity setting, will be active if set to a non-null value:
# verbose=""

# This creates symlinks from the install_tree into the dkms_tree instead of
# copying the modules. This preserves some space on the costs of being less
# safe. Symlinking will be active if set to a non-null value:
# symlink_modules=""
[... 123 lines cut ...]
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.0-2-amd64/updates/dkms/

nvidia-current-peermem.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.0-2-amd64/updates/dkms/
depmod...
Setting up nvidia-driver (510.85.02-5) ...
Processing triggers for glx-alternative-nvidia (1.2.1) ...
Processing triggers for glx-alternative-mesa (1.2.1) ...
Processing triggers for update-glx (1.2.1) ...
Processing triggers for libc-bin (2.35-3) ...
Processing triggers for glx-alternative-nvidia (1.2.1) ...
Processing triggers for libc-bin (2.35-3) ...
Processing triggers for initramfs-tools (0.142) ...
update-initramfs: Generating /boot/initrd.img-5.19.0-2-amd64
```

</details>

xuzhen, 2022-10-27:
You can use a similar helper script. `sign_file=/etc/dkms/sign_helper.sh`
The /etc/dkms/sign_helper.sh is 
```
#!/bin/bash

MOK_PASS=$(keyring get uefi mok)

if [[ $? -ne 0 ]]; then
	keyring set uefi mok
	MOK_PASS=$(keyring get uefi mok)
fi

env KBUILD_SIGN_PIN=$MOK_PASS /lib/modules/"$kernelver"/build/scripts/sign-file "$1" "$2" "$3" "$4"
```

siddhpant, 2022-10-27:
Thanks! But now I am getting

```
Usage: scripts/sign-file [-dp] <hash algo> <key> <x509> <module> [<dest>]
       scripts/sign-file -s <raw sig> <hash algo> <x509> <module> [<dest>]
/etc/dkms/sign_helper.sh: line 14: /var/lib/shim-signed/mok/MOK.priv: Permission denied
/etc/dkms/sign_helper.sh: line 15: /var/lib/shim-signed/mok/MOK.der: Permission denied
/etc/dkms/sign_helper.sh: line 16: /var/lib/dkms/nvidia-current/510.85.02/build/nvidia.ko: Permission denied
```

xuzhen, 2022-10-27:
I guess you forgot the `\` at the end of line 13~15.

siddhpant, 2022-10-27:
oof indeed, there was a novice error (had added comment). Really sorry for the noise.

Thanks once again!

closed 2022-10-27 by siddhpant