dkms-project/dkms #288: Debian users should be able to sign modules with third-party kernels too

mavoga, 2022-12-04:
The `prepare_signing()` function sets the `sign_file` variable this way:
```
sign_file="/usr/lib/linux-kbuild-${kernelver%.*}/scripts/sign-file"
```
but a `sign-file` binary exists in that path only for Debian original kernels. Third party kernels such as Xanmod and Liquorix don't provide a _linux-kbuild-${kernelver%.*}_ package in order to avoid clashing with Debian kbuild packages and include the signing tool in the _headers_ package; with such setting the modules built for such kernels will be signed with the `sign-tool` binary of the Debian original kernel - which can well be a different version - if it's installed on the system, or will fail signing altogether if it's not installed.

The original Debian _linux-headers_ packages always hard depend on the version-matching _linux-kbuild-${kernelver%.*}_ package and a symlink to `/usr/lib/linux-kbuild-${kernelver%.*}/scripts` is always provided as `/usr/src/linux-headers-$kernelver/scripts`; on the other hand it is not possible to compile a kernel module when the headers package is not installed.

For Debian the `sign_file` variable should therefore be set like this in order to correctly sign the modules in every possible configuration:
```
sign_file="/usr/src/linux-headers-$kernelver/scripts/sign-file"
```

Thanks

mavoga, 2022-12-04:
Even better, for cosmetic reasons (the path is printed by dkms):
```
sign_file=$(readlink -f "/usr/src/linux-headers-$kernelver/scripts/sign-file")
```

xuzhen, 2022-12-04:
This issue was already resolved in latest source code.

 `/lib/modules/$kernelver/build/scripts/sign-file` will be used if `/usr/lib/linux-kbuild-${kernelver%.*}/scripts/sign-file` was not found.
https://github.com/dell/dkms/blob/5d343357c339d68b19de8c1f025849700ffb1efe/dkms.in#L881-L895

And you can also set `sign_file="/usr/src/linux-headers-$kernelver/scripts/sign-file"` in /etc/dkms/framework.conf

mavoga, 2022-12-04:
Hi, the issues here are:

1. when the original kernel _is installed_ and the `sign-file` binary _is found_: the modules built for the third-party kernel will be signed with the binary of the original kernel;

2. AFAICS `/etc/dkms/framework.conf` is sourced and `$kernelver` can't be expanded there (at least it was so some time ago when I tried to correct this in the configuration files, don't actually know if this has changed).

xuzhen, 2022-12-04:
>     1. when the original kernel _is installed_ and the `sign-file` binary _is found_: the modules built for the third-party kernel will be signed with the binary of the original kernel;

It's not a problem. sign-file is not affected by any kernel configurations. https://github.com/torvalds/linux/blob/master/scripts/sign-file.c 
I also don't think any 3rd party patches will make the original one unusable.

>     2. AFAICS `/etc/dkms/framework.conf` is sourced and `$kernelver` can't be expanded there (at least it was so some time ago when I tried to correct this in the configuration files, don't actually know if this has changed).

Variable expansion was supported in commit https://github.com/dell/dkms/commit/472774454c53a83f63d20f22f3451ace806e2103

mavoga, 2022-12-04:
Thank you @xuzhen. Ok, got it: Debian has v3.0.8 which don't include neither the fallback `sign_file` setting nor `$kernelver` expansion in conffiles.

I'd rather prefer my solution as it looks simpler and cleaner to me, but it'll definitely work with the patches you introduced, and I'll still have the chance to set the `sign_file` I prefer in a conffile.

Sorry for the noise, closing the issue.

closed 2022-12-04 by mavoga