dkms-project/dkms #289: No way to set KBUILD_SIGN_PIN to unlock module signing key

mavoga, 2022-12-24:
Until v3.0.8 dkms simply sourced configuration files and it was possible to store the key passphrase there in the KBUILD_SIGN_PIN environment variable. Since v3.0.9 the `safe_source()` function has been introduced which prevents this.

How are we supposed to pass the key passphrase to `sign-file`?

Thanks

xuzhen, 2022-12-24:
You can set it in a script: https://github.com/dell/dkms/issues/273#issuecomment-1293431889

mavoga, 2022-12-24:
In a text console:

```
# keyring set uefi mok
Password for 'mok' in 'uefi': 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/keyring/backends/libsecret.py", line 85, in set_password
    stored = Secret.password_store_sync(
gi.repository.GLib.GError: g-io-error-quark: Cannot autolaunch D-Bus without X11 $DISPLAY (0)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/bin/keyring", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3/dist-packages/keyring/cli.py", line 132, in main
    return cli.run(argv)
  File "/usr/lib/python3/dist-packages/keyring/cli.py", line 68, in run
    return method()
  File "/usr/lib/python3/dist-packages/keyring/cli.py", line 85, in do_set
    set_password(self.service, self.username, password)
  File "/usr/lib/python3/dist-packages/keyring/core.py", line 60, in set_password
    get_keyring().set_password(service_name, username, password)
  File "/usr/lib/python3/dist-packages/keyring/backends/libsecret.py", line 94, in set_password
    raise KeyringLocked("Failed to unlock the collection!") from error
keyring.errors.KeyringLocked: Failed to unlock the collection!
```

`keyring` needs X11

mavoga, 2022-12-24:
Furthermore, one has to manually define the `sign_file` variable: the logic for setting it is completely bypassed. This easily brings the user to make a mess.

Can't the passphrase be simply stored in a root only accessible configuration snippet under `/etc/dkms/framework.conf.d/`? This would certainly be sub-optimal from a security point of view, but still better than having the key without passphrase.

xuzhen, 2022-12-24:
> `keyring` needs X11

That script is just an example for the OP of #273. You don't really need `keyring` if you are not him.

siddhpant, 2022-12-24:
> `keyring` needs X11

Yeah, that was a script I personally use (linked in the issue), hence xuzhen's comment there.

You can either [configure it for headless use](https://github.com/jaraco/keyring#using-keyring-on-headless-linux-systems), or just get passphrase through standard ways like `read` everytime, or make a temporary file and storing passphrase in it, and then remembering to delete it later. Something like (I did not test this):

```bash
[...]

MOK_PASS_FILE="/tmp/mok_pass_file"

if [[ ! -f "$MOK_PASS_FILE" ]]; then
	# R/W on tty get around DKMS' possible redirection to /dev/null.
	# Idea from: https://gist.github.com/sbueringer/bd8cec239c44d66967cf307d808f10c4#file-sign-modules-L14
	echo -n "Enter MOK passphrase: " > /dev/tty
	read -s KBUILD_SIGN_PIN < /dev/tty

	# Create file which can be read only by root.
	touch "$MOK_PASS_FILE"
	chmod 600 "$MOK_PASS_FILE"
	sudo chown root:root "$MOK_PASS_FILE"

	# Save passphrase in the file.
	echo "$KBUILD_SIGN_PIN" | sudo tee "$MOK_PASS_FILE" > /dev/null
else
	KBUILD_SIGN_PIN="$(sudo cat $MOK_PASS_FILE)"
fi

[...]
```

(I happened to lurk just by chance, checking if there was a new release.)

mavoga, 2025-04-25:
All this is unnecessarily complicated and requires rewriting most of the logic - and the security measures - already present in dkms.

Since dkms is a well maintained and ubiquitous tool, my humble opinion is that the risks brought by allowing the user to specify a custom sign script can be by far more important compared to those introduced by adding a simple variable to contain the passphrase (and you already have all the logic to read such variables safely in your code).