dkms-project/dkms #290: module version check causes WARNING build vs. install

celane, 2022-12-29:
The changed module version code in v3.0.9 returns mismatched versions on the same file.  (Alma 8.6, but there are reports from other  distros with the same problem).

The source of the problem:
get_module_verinfo() 
   echo ""  (since there is no 'version' line in modinfo)
   echo "A8869EFB63B2CF973E19263"   (srcversion hex string)

compare_module_version reads these two lines into an array for each file, but the test:
if [[ ! "$ver1" ]] || [[ ! "$ver2" ]]; then
        echo "?"
        return 1
    fi

echos "?" with return status of "no match" , because the  [[ ! "$ver1" ]]  test results in 'true' for the blank first entry in the array.

Suggested fix to test: 
if [[  "${#ver1[*]}" == 0 ]] || [[  "${#ver2[*]}" == 0 ]]; then

that checks that the arrays have some entries.

xuzhen, 2022-12-29:
I've opened a pull request for this. But all modules should use `MODULE_VERSION()` in their source code to set a version IMHO

celane, 2022-12-29:
"But all modules should use MODULE_VERSION() in their source code to set a version IMHO"

Strong agreement. 
Unfortunately, spot checking on Alma 8.6 and Fedora 35 didn't turn up *any* installed modules that had a 'Version' set. 
I ran into this problem of modules not having a way to check their version, years ago, and it was frustrating. 

Time to poke at module code, to get a MODULE_VERSION() added.

PR #291 (2023-01-03): Fix warnings for modules with no version
If there is no `MODULE_VERSION()` macro used in a module's source code, `modinfo` will not show the `version: xxxx` field for the compiled binary. 

if `MODULE_VERSION("")` is used, the situation is similar. `modinfo` will only show `version:` for the compiled binary. 

In both cases, `dkms status` will report `WARNING! Diff between built and installed module!` and `dkms uninstall` will say `Module was not found within /lib/modules/$kernelver/` and the the module will not be removed.

Fix #290
`
closed 2023-01-03 by scaronni