dkms-project/dkms #299: Workaround for activating rhel's software collection toolchain

mefff, 2023-01-26:
On Oracle Linux 8, with its uek kernel (a newer kernel they distribute besides the rhel8 one) dkms will fail to build a driver due to trying to build it using an old gcc, with incompatible flags required by the newer kernel:

```
[vagrant@oracle8 ~]$ sudo dkms build -m example-dkms -v 1.0 
Sign command: /lib/modules/5.15.0-6.80.3.1.el8uek.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...
Key file /var/lib/dkms/mok.key not found and can't be generated, modules won't be signed

Building module:
Cleaning build area...
make -j2 KERNELRELEASE=5.15.0-6.80.3.1.el8uek.x86_64 all KVER=5.15.0-6.80.3.1.el8uek.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.0-6.80.3.1.el8uek.x86_64 (x86_64)
Consult /var/lib/dkms/example-dkms/1.0/build/make.log for more information.
[vagrant@oracle8 ~]$ cat /var/lib/dkms/example-dkms/1.0/build/make.log 
DKMS make.log for example-dkms-1.0 for kernel 5.15.0-6.80.3.1.el8uek.x86_64 (x86_64)
Thu Jan 26 14:17:36 UTC 2023
make -C /lib/modules/5.15.0-6.80.3.1.el8uek.x86_64/build M=/var/lib/dkms/example-dkms/1.0/build modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/kernels/5.15.0-6.80.3.1.el8uek.x86_64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 11.2.1 20220127 (Red Hat 11.2.1-9.1.0.2)
  You are using:           gcc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-16.0.2)
  CC [M]  /var/lib/dkms/example-dkms/1.0/build/driver.o
gcc: error: unrecognized command line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command line option ‘-fzero-call-used-regs=used-gpr’
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/example-dkms/1.0/build/driver.o] Error 1
make[1]: *** [Makefile:1976: /var/lib/dkms/example-dkms/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.15.0-6.80.3.1.el8uek.x86_64'
make: *** [Makefile:4: all] Error 2
```

This is with a hello world kernel module and this dkms.conf

```
PACKAGE_NAME="example-dkms"
PACKAGE_VERSION="1.0"
CLEAN="make clean"
MAKE[0]="make all KVER=${kernelver}"
BUILT_MODULE_NAME[0]="example-dkms"
DEST_MODULE_LOCATION[0]="/updates/dkms"
AUTOINSTALL="yes"
```

This could be solved by building it with: `scl enable gcc-toolset-11 -- dkms install -m <driver>/<version> --force`, but that hard to integrate into the package manager. For example, if the dkms module was packaged in an rpm, doing `scl enable gcc-toolset-11 -- rpm -i example-dkms.rpm` would also fail, probably because the variables that `scl` set are not set when dkms runs.

The `scl enable` command is basically setting some environment variables, extending the `$PATH` pointing to the newer binaries, etc. Is there other alternative to make dkms aware of this new tools?

mefff, 2023-01-26:
When running with `scl enable gcc-toolset-11 -- dkms build example-dkms/1.0` it does, but when trying to install the package with rpm it doesn't, even with `scl enable gcc-toolset-11 -- rpm -i example-dkms.rpm`. Is like rpm is changing the PATH somehow, or not exporting it to the installation scripts.

So, when dkms has both gcc on PATH it finds the correct one, I presume because the newer is first on PATH? probably with #298 will get it always, besides the PATH order

xuzhen, 2023-01-26:
I see. With #298, you can create a little script to help dkms choose the right toolchain.
```
#!/bin/sh
. /opt/rh/gcc-toolset-11/enable
gcc "$@"
```
Save it to `/usr/local/bin/gcc-11`, and `chmod +x /usr/local/bin/gcc-11`.

scaronni, 2025-01-15:
This should be solved by adding a `PRE_BUILD` directive in the `dkms.conf` file of the module. Example:
```
PRE_BUILD"scl enable gcc-toolset-11"
```
Or calling a specific script shipped along with the module in the `PRE_BUILD` directive.

Anyway, the specific Red Hat Software Collection enablement mechanism should not be in DKMS.

closed 2025-01-15 by scaronni