dkms-project/dkms #301: DKMS seems to erroneously apply `--no-initramfs` whe uninstalling or installing modules

BrianAllred, 2023-02-01:
On Tumbleweed, installing the openrazer drivers from [the official repository](https://build.opensuse.org/project/show/hardware:razer) results in the attached zypper log. The openrazer maintainer says the RPM spec files don't use `--no-initramfs`, and upon inspection, he seems to be correct.

These drivers installed and worked for me before and only broke sometime in the past week or two.

[razer.txt](https://github.com/dell/dkms/files/10560243/razer.txt)

luc14n0, 2023-02-01:
Please add the version of **DKMS** that was used.

I suppose that the interesting part of the log is:

```
# Removing any linked weak-modules
# getopt: unrecognized option '--no-initramfs'
# Usage: ${0##*/} [options] {--add-modules|--remove-modules}
#        ${0##*/} [options] {--add-kernel|--remove-kernel} {kernel-release}
# ...
```

And if the logic of the **IF** statement that adds `--no-initramfs` in the code is right, we may have to knock on [openSUSE/suse-module-tools](https://github.com/openSUSE/suse-module-tools)'s door instead, the repo responsible for **openSUSE**'s `weak-module` script.

BrianAllred, 2023-02-06:
So I'm not sure what fixed it, but it seems to be fixed after the latest rounds of Tumbleweed updates.

closed 2023-02-06 by BrianAllred