dkms-project/dkms #302: dkms autoinstall fails on modules not built due to BUILD_EXCLUSIVE_*

anbe42, 2023-02-06:
I've edited dkms.conf to contain
```
...
AUTOINSTALL="yes"
BUILD_EXCLUSIVE_ARCH="none"
```
resulting in this error:
```
# dkms autoinstall -k 6.1.0-3-amd64 ; echo $?
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Error! The /var/lib/dkms/bbswitch/0.8/6.1.0-3-amd64/x86_64/dkms.conf for module bbswitch includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch/config.
This indicates that it should not be built.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
11
```

xuzhen, 2023-02-06:
This is exactly what `BUILD_EXCLUSIVE_ARCH` does.
```
BUILD_EXCLUSIVE_KERNEL=
    This optional directive allows you to specify a regular expression which defines the subset of kernels which DKMS is  allowed  to  build  your module  for.  If the kernel being built for does not match against this regular expression, the dkms build will error out. For example, if you set it as ="^2.4.*", your module would not be built for 2.6 kernels.

BUILD_EXCLUSIVE_ARCH=
    This optional directive functions very similarly to BUILD_EXCLUSIVE_KERNEL except that it matches against the kernel architecture.  For  example, if you set it to ="i.86", your module would not be built for ia32e, x86_64, amd64, s390, etc.
```

anbe42, 2023-02-06:
Yes, but why does it cause dkms autoinstall to fail?

anbe42, 2023-02-07:
@evelikov I'd expect `dkms autoinstall` to exit successfully if there is nothing to do.

evelikov, 2023-02-07:
Tests or it didn't happen :-P

xuzhen, 2023-02-07:
Under normal circumstances, DKMS will only process one module in one run. Therefore, the way it handles the error is generally exited directly.

However, the autoinstall function will process multiple modules, so once there is a problem within any module, the entire process is interrupted.

xuzhen, 2023-02-07:
https://github.com/dell/dkms/blob/ce3e217f7d3746ab23cedf86613f636780b23a72/dkms.in#L2181-L2185

If we want to keep the `autoinstall()` in `dkms.in`, we have to modify almost everywhere where the `die()` appears

closed 2023-04-03 by evelikov

cregory777, 2023-08-09:
no matter what i do this is the output's that i get 

Setting up linux-image-6.3.0-2-amd64 (6.3.11-1) ...
/etc/kernel/postinst.d/dkms:

dkms: autoinstall for kernel 6.3.0-2-amd64 was skipped since the kernel headers for this kernel do not seem to be installed.

update-initramfs: Generating /boot/initrd.img-6.3.0-2-amd64
grep: /boot/config-6.3.0-2-amd64: No such file or directory
W: zstd compression (CONFIG_RD_ZSTD) not supported by kernel, using gzip
grep: /boot/config-6.3.0-2-amd64: No such file or directory
E: gzip compression (CONFIG_RD_GZIP) not supported by kernel
update-initramfs: failed for /boot/initrd.img-6.3.0-2-amd64 with 1.
run-parts: /etc/kernel/postinst.d/initramfs-tools exited with return code 1
dpkg: error processing package linux-image-6.3.0-2-amd64 (--configure):
 installed linux-image-6.3.0-2-amd64 package post-installation script subprocess returned error exit status 1

 dpkg: dependency problems prevent configuration of linux-image-amd64:
 linux-image-amd64 depends on linux-image-6.3.0-2-amd64 (= 6.3.11-1); however:
  Package linux-image-6.3.0-2-amd64 is not configured yet.

dpkg: error processing package linux-image-amd64 (--configure):
 dependency problems - leaving unconfigured
Setting up ibus-data (1.5.29~beta1-3) ...
Setting up libwxbase3.2-1:amd64 (3.2.2+dfsg-3) ...
Setting up python3-sphinxcontrib.jquery (4.1-3) ...
Setting up eject (2.39.1-3) ...
Setting up grub-pc (2.06-13+kali1) ...
sudo: update: command not found
dpkg: error processing package grub-pc (--configure):
 installed grub-pc package post-installation script subprocess returned error exit status 1

 insserv: Script `ssh' has overlapping Default-Start and Default-Stop runlevels (2 3 4 5) and (2 3 4 5). This should be fixed.

 update-initramfs: Generating /boot/initrd.img-6.1.0-kali9-amd64
W: Possible missing firmware /lib/firmware/amdgpu/ip_discovery.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/vega10_cap.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/sienna_cichlid_cap.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_cap.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/psp_13_0_10_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/psp_13_0_10_sos.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_cap.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_3_imu.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_3_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_3_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_3_me.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_3_pfp.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_0_toc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/sdma_6_0_3.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/sienna_cichlid_mes1.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/sienna_cichlid_mes.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi10_mes.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_3_mes1.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/gc_11_0_3_mes.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/smu_13_0_10.bin for module amdgpu

No such key “background-transparency-percent” in schema “org.gnome.Terminal.Legacy.Profile” as specified in override file “/usr/share/glib-2.0/schemas/21_kali-themes.gschema.override”; ignoring override for this key.
No such key “use-transparent-background” in schema “org.gnome.Terminal.Legacy.Profile” as specified in override file “/usr/share/glib-2.0/schemas/21_kali-themes.gschema.override”; ignoring override for this key.

Errors were encountered while processing:
 linux-image-6.3.0-2-amd64
 linux-image-amd64
 grub-pc
needrestart is being skipped since dpkg has failed

dpkg: error processing package linux-image-amd64 (--configure):
 dependency problems - leaving unconfigured
Setting up grub-pc (2.06-13+kali1) ...
sudo: update: command not found
dpkg: error processing package grub-pc (--configure):
 installed grub-pc package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 linux-image-6.3.0-2-amd64
 linux-image-amd64
 grub-pc
needrestart is being skipped since dpkg has failed
E: Sub-process /usr/bin/dpkg returned an error code (1)

anbe42, 2023-08-09:
The last comment is unrelated to this bug report.
The `linux-image-6.3.0-2-amd64` package is not installed, but you should rather switch to a 6.4 kernel (and remove 6.3.0-2 afterwards), as 6.3.0-2 exposed a packaging bug that has been fixed with 6.4.

evelikov, 2023-08-09:
@cregory777 for the future please open new issue with clear reproduction steps and environment details.

commit 0f499c98 (2023-04-03): autoinstall: do not fail on modules skipped due to BUILD_EXCLUSIVE_*

Closes #302