dkms-project/dkms #303: Why /etc/dkms/<module>.conf seems ignored?

AGI-chandler, 2023-02-06:
Hi I would post a question in the discussions but is it disabled?&nbsp;  I have Debian's stable version 2.8.4-3 installed and it seems to ignore my `/etc/dkms/zfs.conf` in which I have just a single line to set an environment variable: `ICP_ROOT="/usr/local/src/QAT.L.4.20.0-00001"`.&nbsp; However, when I install `zfs-dkms` package it isn't picking up this variable and thus the module gets installed incorrectly.&nbsp; If I instead export the variable to my shell or even just run `ICP_ROOT="/usr/local/src/QAT.L.4.20.0-00001" apt install --reinstall zfs-dkms` then it picks up the variable and properly installs the module.&nbsp; I don't understand why the setting in `/etc/dkms/zfs.conf` has no effect then, any ideas?&nbsp; Thanks so much

evelikov, 2023-02-07:
Looking at [the package](https://packages.debian.org/bullseye/all/zfs-dkms/filelist) - there is no `/etc` file provided by default. Can you attach the one you're using?

In general only a set of variables are allowed to be set in the configuration files. Once we have a look at the file itself, we can provide you with some concrete suggestions.

xuzhen, 2023-02-07:
Since the `PRE_BUILD` script in `/usr/src/zfs-2.1.9/dkms.conf` source in `/etc/dkms/zfs.conf`, you can make `/etc/dkms/zfs.conf` be like this:
```
echo --with-qat=/usr/local/src/QAT.L.4.20.0-00001
```

AGI-chandler, 2023-02-11:
Hi @evelikov it just has the one line like I mentioned, so:
```
# cat /etc/dkms/zfs.conf 
ICP_ROOT="/usr/local/src/QAT.L.4.20.0-00001"
# 
```
My other bud who also uses ZFS+QAT on Debian said this worked for them so that's where I got it from.

Thanks @xuzhen I will try that instead.

AGI-chandler, 2023-02-20:
@xuzhen That worked!
Thanks so much!

closed 2023-02-20 by AGI-chandler