dkms-project/dkms #306: consistent error-codes

umlaeute, 2023-02-20:
while investigating mitigations for #302, i thought about selectively ignoring some exit-codes in our dkms-wrappers...
only to discover, that the exit-codes are just a wild-bunch of numbers without any apparent systematic.

Here's a list of how `dkms` currently dies:

|err| message |
|----|----|
| 0 | *Remove cancelled because --rpm_safe_upgrade scenario detected.* |
| 1 | *Unable to make temporary file.* |
| 1 | *If more than one arch is specified on the command line, then there must be an equal number of kernel versions also specified (1:1 relationship).* |
| 1 | *Arguments <module> and <module-version> are not specified.* |
| 1 | *Invalid number of parameters passed.* |
| 1 | *No write access to DKMS tree at ${dkms_tree}* |
| 1 | *Unable to make temporary directory* |
| 1 | *You must be root to use this command with the --force option.* |
| 1 | *You must be root to use this command.* |
| 1 | *Your kernel headers for kernel $1 cannot be found at /lib/modules/$1/build or /lib/modules/$1/source.* |
| 2 | *You cannot specify a kernel version and also specify "--all" on the command line.* |
| 2 | *$archive_location does not exist.* |
| 2 | *Could not find module source directory.* |
| 2 | *The templatekernel and the specified kernel version are the same.* |
| 2 | *Unknown option: $1* |
| 3 | *You cannot specify an arch and also specify "--all" on the command line.* |
| 3 | *DKMS tree already contains: $module-$module_version* |
| 3 | *Tarball does not appear to be a correctly formed DKMS archive. No dkms.conf found within it.* |
| 3 | *The module/version combo: $module-$module_version is not located in the DKMS tree.* |
| 3 | *The module: $module is not located in the DKMS tree.* |
| 3 | *This module/version has already been built on: $kernelver* |
| 4 | *Cannot specify more than one action.* |
| 4 | *Could not locate dkms.conf file.* |
| 4 | *The action $1 does not support multiple kernel version parameters on the command line.* |
| 5 | *The action $1 does not support the --all parameter.* |
| 5 | *This module/version combo is already installed for kernel $kernelver ($arch).* |
| 6 | *No modules built for ${kernelver[$i]} (${arch[$i]}).* |
| 6 | *Failed to make tarball.* |
| 6 | *Installation aborted.* |
| 6 | *Problems with depmod detected. Automatically uninstalling this module.* |
| 6 | *The directory $install_tree/$kernelver doesn't exist.* |
| 7 | *No valid dkms.conf in dkms_source_tree or dkms_binaries_only.* |
| 8 | *$PACKAGE_NAME-$PACKAGE_VERSION is already added!* |
| 8 | *You have specified both --binaries-only and --source-only.* |
| 8 | *Bad conf file.* |
| 8 | *The directory $source_dir does not appear to have module source located within it.* |
| 9 | *$1 must contain a dkms.conf file!* |
| 9 | *Missing write permissions for $tarball_dest.* |
| 9 | *The $base_dir/dkms.conf for module $module includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch.* |
| 9 | *Unable to install $archive_location using rpm.* |
| 10 | *Malformed dkms.conf file. Cannot load source tree.* |
| 11 | *One or more modules failed to install during autoinstall.* |
| 101 | *pre_install failed, aborting install.* |

as you can see, exit codes are re-used multiple times for unrelated issues as well as related issues use different exit codes.
e.g. `1` could mean any of:
- invalid cmdline arguments
- missing files/directories
- missing generic privileges ("must be root")
- missing specific priviliges ("no write access to ...")

otoh, `missing write permissions` is covered by:
- `1` ("No write access to DKMS tree")
- `9` ("Missing write permissions for $tarball_dest.")

while the latter probably makes sense (as it allows one to nail down the specific error), the former does not make sense at all (at least: to me).

i have no idea whether the exit-codes are part of any interface, but i would suggest to
1. use named error-codes (which should make it easier to pick the right one when adding a new `die` stanza)
2. group error-codes by functionality (e.g. have all "wrong cmdline arguments" errors emit the same status-code (or a similar; but distinct from unrelated errors)
3. probably base the exit-codes on some established standards, as documented in `errcode.h` or `sysexits.h`
  - http://mazack.org/unix/errno.php
  - https://tldp.org/LDP/abs/html/exitcodes.html
  - https://stackoverflow.com/questions/1101957/

xnox, 2023-03-09:
On Ubuntu, we rely on `9 - The $base_dir/dkms.conf for module $module includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch.` and ideally it would be kept at just that. It is unfortunate that `9 - $1 must contain a dkms.conf file!` exists, as that's very different from build exclusive return code. Missing dkms.conf is fatal imho, whereas build exclusive is semantic and different actions can be taken based off that.

Separately, it would be useful for us to check some other way if a given dkms is compatible with a given kernel. I.e. just run the compatiblity check on exclusives with a new subcommand. Such that we don't have to rely on error code 9, or like match stderr strings.

evelikov, 2023-03-09:
@xnox thanks for the valuable reference points. If you'd like to add tests or propose something concrete (ideally backed with tests) I would gladly review and try to get it merged.

xnox, 2023-04-05:
Note that code 9 got changed to 77 in master since previously mentioning that 9 is being relied upon in https://github.com/dell/dkms/commit/cac91325f4f9a10055a2d5b414808065cd640098

This got synced into Ubuntu post feature & kernel freeze, and will now require change to the Ubuntu kernel for Lunar release.

cc: @evelikov @anbe42 @vorlonofportland @LocutusOfBorg

evelikov, 2023-04-05:
I'm not sure I follow. What does "this" refer to in "This got synced into Ubuntu ..." 

Note that the change isn't part of a dkms release, just yet. So if the team has preferences or suggestions we have time to get those in.

evelikov, 2023-04-05:
Seems like bunch of the fixes got backported to Debian (and Ubuntu?). Most notable being [the 9 -> 77 change]( https://salsa.debian.org/debian/dkms/-/commit/9b3fb045db3069f529284f77b763b94a158adab1).

Speaking as a casual user of Debian/Ubuntu with zero knowledge of their processes: having a functional/behavioural change like this in a `3.10-7` -> `3.10-8` package release does not seem wise. Users expect things to _not_ change, let alone in such subtle ways. IMHO the patch should be deferred to the `3.11-x` releases.

LocutusOfBorg, 2023-04-05:
Hello, dkms won't migrate into lunar, unless we force it.
Lunar-proposed is broken, lunar release is fine.

xnox, 2023-04-05:
> Seems like bunch of the fixes got backported to Debian (and Ubuntu?). Most notable being [the 9 -> 77 [...]

indeed, i requested multiple times for @LocutusOfBorg to stop uploading breaking changes to dkms that break ubuntu kernel builds without any testing or coordination. And yet again it has happened. at this point i will raise the actions with the techboard, requesting removal upload rights of the dkms package in ubuntu, because it severily impacted development of ubuntu - jeopardizing release schedule multiple times.

LocutusOfBorg, 2023-04-05:
Hello @xnox that sync was coordinated with release team, but I assume all the responsibility for not understanding this subtle change. I'm sorry for that, and will make sure it won't migrate in release. I was never told that the kernel relied on such return code, and my bad for not understanding it before (and now that you pointed it, I should have understood it by myself). That said, dkms is in proposed only pocket, and won't migrate in release, so I don't understand why we should patch the kernel for lunar...

xnox, 2023-04-05:
> Hello @xnox that sync was coordinated with release team, but I assume all the responsibility for not understanding this subtle change. I'm sorry for that, and will make sure it won't migrate in [...]

src:linux build-depends on dkms and invokes dkms during build. src:linux in lunar is build in lunar-proposed, but released in lunar (release). all packages must build correctly in lunar (release). Thus src:linux must build correctly against either dkms packages that exist in lunar-proposed and lunar (release).

Your lack of understanding here, after causing kernel team to not able to build new kernels this cycle multiple times, further indicates that you should stop uploading dkms. You have caused multiple weeks of delays in packaging new kernel this cycle in Ubuntu.

When uploading new packages, in general one should inspect, test or at least consider and understand reverse build dependencies as listed in `reverse-depends -b dkms`.

evelikov, 2023-08-15:
AFAICT the most important error code is 77 aka skip. It was introduced recently alongside a bunch of tests - huge thanks to Andreas.

I believe we can close this issue, for now. If there's a particular need to fix a specific one MR are greatly appreciated - please don't forget to do some homework, to ensure existing users don't regress (too badly).

closed 2023-08-15 by evelikov