dkms-project/dkms #326: Kernel 5.15 Lts Fedora Copr : "DKMS will not replace this module."

enigma131, 2023-04-01:
Hi,

After signature solved problem, I have a new little issue with this kernel
After analysing the source code of dkms.in, it seems it looks for comparing versions before inserting.
The Copr kernel has no version, so comparing gives identical binarys.
Example : 
modinfo stb6100
filename:       /lib/modules/5.15.105-200.fc37.x86_64/extra/stb6100.ko.xz
license:        GPL
description:    STB6100 Silicon tuner
author:         Manu Abraham, Enigma13
depends:        
retpoline:      Y
name:           stb6100
vermagic:       5.15.105-200.fc37.x86_64 SMP mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        30:6A:97:EF:29:57:61:5F:1C:3C:78:2E:A7:94:56:8E:91:D3:67
sig_hashalgo:   sha512
signature:      34:94:EA:27:5B:AE:48:8A:69:A6:90:4F:D3:95:7E:2E:FB:61:25:50:
		02:38:13:3D:23:7B:5D:91:7D:CE:69:B6:D1:28:B5:D8:A0:36:11:CE:

For Redhat: 
modinfo stb6100
filename:       /lib/modules/4.18.0-425.13.1.el8_7.x86_64/extra/stb6100.ko.xz
license:        GPL
description:    STB6100 Silicon tuner
author:         Manu Abraham, Enigma13
rhelversion:    8.7                                                   <-----------------
srcversion:     E29A9BB9B806FD74F273D68     <-----------------
depends:        
name:           stb6100
vermagic:       4.18.0-425.13.1.el8_7.x86_64 SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        30:6A:97:EF:29:57:61:5F:1C:3C:78:2E:A7:94:56:8E:91:D3:67
sig_hashalgo:   sha256
signature:      57:0C:A1:A0:7F:C5:D5:D1:17:56:B4:57:15:CC:90:E7:63:60:C0:8F:
		08:71:AE:EA:05:93:83:83:05:00:83:28:6B:DB:7A:03:62:1A:7E:02:

How to solve this ?
Manually, after each kernel upgrade, i've done manually : 

sudo dkms autoinstall --verbose --force

+ reboot

evelikov, 2023-04-19:
I'm not sure I follow what the issue is. Can you provide some reproduction steps?

There is an issue about our questionable version checking - https://github.com/dell/dkms/issues/311 - which, once untangled could solve your issue.

enigma131, 2023-04-19:
Thanks for answer.
The kernel is not compiled by me, but by the Copr Guy: https://copr.fedorainfracloud.org/coprs/kwizart/kernel-longterm-5.15/packages/
Can I found somewhere a trick to autoinstall via dkms and no src version from that kernel  ?
Because it has no src version, it says it is same version after compiling and didn't auto install.
Compilation phase is ok. 
After compiling and reboot, new modules are not installed, here I have to force manually install via --force argument.
After a second reboot, modules are correctly installed.

evelikov, 2023-04-20:
If you can provide some concrete reproduction steps, then we can add a test to ensure the issue gets fixed and stays fixed. At the very least can you share:

- the .config file for both "good" and "bad" kernel/module
- the complete output of `modinfo` for both "good" and "bad" module
- link to spec file, makefile, etc, for the good case

enigma131, 2023-04-20:
There are no bad and good modules, I repeat the compilation process is OK and all are ok.
The problem is that it detects same VERSION in post stage because of missing srcversion in copr kernel 5.15 compiled module (see my first post), so it didn't auto install.
Look in compare_module_version in source dkms.in .
And later :
            error $"Module version $(get_module_verinfo "${dkms_module}" | head -n 1) for $4${module_suffix}" \
                $"is not newer than what is already found in kernel $1 ($(get_module_verinfo "${kernels_module}" | head -n 1))." \
                $"You may override by specifying --force."