dkms-project/dkms #329: setup_kernels_arches should not call rpm on Debian/Ubuntu based distros

anbe42, 2023-04-07:
If the distribution is based on Debian/Ubuntu, `dkms`  should not make any calls to `rpm`.
There are currently unguarded calls in `setup_kernels_arches` and `do_uninstall`.

There is an unwanted side effect of creating an empty rpmdb in `/.rpmdb/` or `/root/.rpmdb/`  on Debian and derivatives when performing `rpm` queries and `rpm` is installed.

evelikov, 2023-04-19:
Glancing at the two uses:
 - the i32a in `setup_kernels_arches` was added ~15 years ago with https://github.com/dell/dkms/commit/5a9c77b62c553b33fcf1eacfe6a50b71061c7bf9 by @superchalupa.

Michael do you recall what distro/version it's aimed at, is it still relevant?

  - while the `do_uninstall` reference is more recent https://github.com/dell/dkms/commit/79fc8551919623836ebe72cde96ce58316e73a54 by @vt-alt.

What are the side-effects of removing the (empty) folder when it's owned by rpm? Should we do a similar check for other package manageds - yum, apt, pacman, etc? I don't think such a check scales well, plus my gut feeling is that this might be a workaround for broken package.

anbe42, 2023-04-19:
a possible solution: https://github.com/anbe42/dkms/commit/096979dad454892cfb33c6d107fbfc77941bf3d4
this adds a `case "$running_distribution" in` around rpm usage but does not try to reimplement the rpm usage with dkms/pacman, not yet tested ;-)

evelikov, 2023-04-20:
Our ultimate goal is to remove all `running_distribution` instances. Iff needed a config toggle can be used to control the code flow. Considering the side effect is couple of empty directories, I'm not in favour of adding more.

evelikov, 2023-08-15:
I've added the required notes to the "move distro specifics into the config" task and I'll be closing this in favour of it.

Hope that's OK, I see you've got a local workaround in Debian for the short term.

commit 878c6180 (2025-02-04): dkms: do not perform rpm queries on Debian/Ubuntu/Arch based distributions

Closes #329
closed 2023-08-15 by evelikov