dkms-project/dkms #337: Deprecation of MAKE_INITRD

morong, 2023-07-28:
Storage drivers installed via DKMS should automatically be included in the initrd.

I know this change was made back in 2021 but we are just noticing it with DKMS 3.x showing up in distros like the just released Debian 12.  In removing MAKE_INITRD feature, was there consideration made for storage drivers supporting the boot controller?  Those drivers are usually included in the initrd.  This is also important when the DKMS storage driver is displacing an inbox version of the driver.  While I can tell customers that they need to remake their initrd following a DKMS driver install, not sure how that will work during a kernel upgrade scenario.  DKMS will discover a new kernel and rebuild the driver but not included it in the initrd.  DKMS is our only driver packaging available to our customers for Debian and Ubuntu, since those distros do not support a stable KABI.

anbe42, 2023-07-28:
I'm writing from the Debian side, I don't know in detail how these things work in other distros. Even Ubuntu might behave differently if the kernel is involved.

Rebuilding an initrd is a very distribution specific thing and dkms does not need to know how to do that properly everywhere. On Debian, upon kernel installation/upgrade, e.g. the scripts `/etc/kernel/postinst.d/dkms` (from dkms) and `/etc/kernel/postinst.d/initramfs-tools` (from initramfs-tools) take care of building the modules and updating the initrd thereafter, so this should work out-of-the-box on a standard installation.

But rebuilding the initrd does not imply that a new module gets included. The Debian initrd only contains a small subset of modules that may be needed for booting the system.

See e.g. the `r8168` source package for a network driver module that gets included in the initrd. `r8168-dkms` ships `/usr/share/initramfs-tools/hooks/r8168-dkms` and (implicitly) activates the `update-initramfs` trigger to ensure the initramfs gets updated (and includes the r8168.ko module) upon installation of the package.
`zfs-dkms` might be a more complex example (but I haven't checked the package in detail).

I do not know what module you have in mind (or how you distribute it), but I assume it is not available in any distribution and thus has not been exposed to the dkms developers or distribution maintainers.

morong, 2023-07-28:
Hello Andreas,
Thank you for your feedback.  The Microchip driver, smartpqi (https://github.com/Smart-Storage/smartpqi) is the one in question. It is an inbox driver for both Debian and Ubuntu.  The smartpqi DKMS package we provide allows for updating the driver beyond the inbox driver version.   When the DKMS package gets installed, the driver ends up in the initrd automatically, prior to Debian 12 and/or DKMS 2.8.8.
Our continued testing confirms the details you provided above regarding kernel install/upgrade.  We do see the initrd being updated with our driver from DKMS for the new kernel.  That is a huge relieve. 
So, the net of this MAKE_INITRD deprecation, is that customers will need to add the step of "update-initramfs -u" right after they installation of our driver DKMS package.  I can live with that.

closed 2023-07-28 by morong

evelikov, 2023-07-29:
When removing the option I was aware of 3 initrd tools - dracut, initramfs and mkinitcpio. Of which DKMS supported only the former two, rather poorly at that. Since then there are 2 new cool kids - booster and there's a systemd tool (don't recall the name sorry).

As mentioned above - the process, initrd name location/pattern and others varies across distros. While compounding these you can see the combinatoric explosion and how it's not practical/possible have it working.

More importantly, the end user should know and have control if/when the initrd is regenerated - with the guidance of the distro of course.

Searching for smartpqi shows that the driver has been upstream since 2016, so I'm a little confused why you/your customers are using DKMS.

HTH

anbe42, 2023-08-01:
You should be able to automate the initrd regeneration by adding `activate-noawait update-initramfs` to `debian/triggers` in the source package that builds your `whatever-dkms.deb` package. (Do you have the packaging available in a public repository, too?)

morong, 2023-08-02:
Andreas,
Thank you. I will take a look at the debian/triggers option.  Only source and dkms.conf in our github repository.