dkms-project/dkms #338: limiting a modules inclusion

sebastianschauenburg, 2023-07-31:
Good morning everyone,

Current situation: Running AMD's proprietary graphics drivers (using DKMS) on Ubuntu 23.04, but they only have support up to 22.04. So I installed an older kernel and that works. But DKMS tries to build and install ```amdgpu``` for **all** kernels (which fails on the newer kernels) and thus errors occur. Building and installing only works for the older kernel, not the newer ones (which are installed though).

It would be great if I could limit the DKMS build/inclusion to certain kernel versions or exclude certain kernels versions. Could not really find a way for this in the docs, hence this ticket. Any help would be appreciated.

xuzhen, 2023-07-31:
Since the dkms package in Ubuntu 23.04 is version 3.0.10, the `BUILD_EXCLUSIVE_KERNEL` directive could be helpful.

From `man dkms`
```
       BUILD_EXCLUSIVE_KERNEL=
              This  optional  directive  allows  you to specify a regular expression which defines the subset of kernels which DKMS is allowed to build your
              module for.  If the kernel being built for does not match against this regular expression (or does not the  satisfy  the  constraints  of  any
              other  BUILD_EXCLUSIVE_* directive), the dkms build will error out with exit code 77.  Note that dkms autoinstall will ignore this type of er‐
              ror condition and simply skip the respective modules.  For example, if you set it as ="^2.4.*", your module would not  be  built  for  2.6  or
              later kernels.
```

`BUILD_EXCLUSIVE_KERNEL="^5.*"` will make dkms ignore kernel 6.x of Ubuntu 23.04

For dkms 3.0.11+, `BUILD_EXCLUSIVE_KERNEL_MAX` can be used instead.

sebastianschauenburg, 2023-07-31:
Had the man-page open earlier, but totally missed that option. Excellent, thank you so much :-)

closed 2023-07-31 by sebastianschauenburg