dkms-project/dkms #361: DKMS Does Not Sign v4l2loopback-dkms

v4u6h4n, 2023-11-08:
Hey everyone, I'm trying to get DKMS to automatically sign modules, specifically v4l2bookback-dkms for OBS Studio virtual webcam, and I've been at it for many, many fruitless hours, and I am now at a bit of a loss what to do... any suggestions would be appreciated. I have completed the following steps, and even tried them a second time, using `mokutil --delete` beforehand, in case I made some mistake in the first attempt.

Setup Keys
> `sudo mokutil --sb-state`
> SecureBoot enabled
> `sudo apt install dkms`
> `sudo mkdir -p /var/lib/shim-signed/mok/`
> `cd /var/lib/shim-signed/mok/`
> `sudo openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -days 36500 -subj "/CN=Secure Boot Key/"`
> `sudo openssl x509 -inform der -in MOK.der -out MOK.pem`
> `sudo mokutil --import /var/lib/shim-signed/mok/MOK.der`

Reboot and enrol key.

> `sudo mokutil --test-key /var/lib/shim-signed/mok/MOK.der`
> /var/lib/shim-signed/mok/MOK.der is already enrolled

Setup DKMS

> `sudo nano /etc/dkms/framework.conf`
> mok_signing_key="/var/lib/shim-signed/mok/MOK.priv"
> mok_certificate="/var/lib/shim-signed/mok/MOK.der"
> sign_tool="/etc/dkms/sign_helper.sh"

Install v4l2loopback

> `sudo apt install v4l2loopback-dkms`
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Suggested packages:
>   v4l2loopback-utils
> The following NEW packages will be installed:
>   v4l2loopback-dkms
> 0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
> Need to get 37.2 kB of archives.
> After this operation, 118 kB of additional disk space will be used.
> Get:1 http://deb.debian.org/debian unstable/main amd64 v4l2loopback-dkms all 0.12.7-2 [37.2 kB]
> Fetched 37.2 kB in 0s (75.0 kB/s)          
> Selecting previously unselected package v4l2loopback-dkms.
> (Reading database ... 170264 files and directories currently installed.)
> Preparing to unpack .../v4l2loopback-dkms_0.12.7-2_all.deb ...
> Unpacking v4l2loopback-dkms (0.12.7-2) ...
> Setting up v4l2loopback-dkms (0.12.7-2) ...
> Loading new v4l2loopback-0.12.7 DKMS files...
> Building for 6.5.0-4-amd64
> Building initial module for 6.5.0-4-amd64
> At main.c:170:
> - SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
> - SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
> - SSL error:1C80009F:Provider routines::unable to get passphrase: ../providers/implementations/encode_decode/decode_epki2pki.c:96
> - SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
> - SSL error:04800068:PEM routines::bad password read: ../crypto/pem/pem_pkey.c:159
> sign-file: /var/lib/shim-signed/mok/MOK.priv
> Done.
> 
> v4l2loopback.ko:
> Running module version sanity check.
>  - Original module
>    - No original module exists within this kernel
>  - Installation
>    - Installing to /lib/modules/6.5.0-4-amd64/updates/dkms/
> depmod...

Troubleshoot

`sudo nano /etc/dkms/sign_helper.sh`
#!/bin/bash
/lib/modules/"$1"/build/scripts/sign-file sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der "$2"

`sudo apt purge v4l2loopback-dkms`
`sudo apt install v4l2loopback-dkms`

Same errors as the first time. What am I doing wrong? It looks like it's saying the PEM passphrase is wrong or something? Not sure... Oh and my system, in case it's relevant.

Debian Unstable (also tried the same steps in Debian 12 yesterday)

v4u6h4n, 2023-11-09:
I have replicated the SSL error when attempting to [manually build](https://github.com/umlaeute/v4l2loopback#dkms) the module with DKMS.

anbe42, 2023-11-24:
This bug is unrelated to the actual module, the "problem" is that you are using a key that is protected by a passphrase (which is a good thing), but dkms makes no attempt to supply a passphrase (or somehow prompt for it).

v4u6h4n, 2023-11-24:
> This bug is unrelated to the actual module, the "problem" is that you are using a key that is protected by a passphrase (which is a good thing), but dkms makes no attempt to supply a passphrase (or somehow prompt for it).

Hey :-) Oh so DKMS doesn't support key passphrases yet? If I leave the passphrase blank when creating the key will that circumvent the bug? I would prefer to have a passphrase, but at this point I have secure boot disabled because I need that module up and running, so it would still be an improvement.

xuzhen, 2023-11-25:
> Oh so DKMS doesn't support key passphrases yet?

You can set the KBUILD_SIGN_PIN environment variable in a helper script https://gist.github.com/siddhpant/19c07b07d912811f5a4b2893ca706c99

anbe42, 2023-11-29:
But at least dkms should document that
 a) the signing key created by dkms has no passphrase
 b) the user needs to do additional steps if he wants to use a custom key that is protected by a passphrase (and give a pointer where to find additional instructions), maybe describe one possible solution (simple (manually providing the passphrase on the command line), not elegant ((automatically) providing the passphrase in some keyring))

v4u6h4n, 2023-11-30:
Hey everyone, thanks for the suggestions, the kernel module in question is running on my production machine, so it's a bit tricky testing out the suggested workarounds, but will try and get it done this week. I think v4l2loopback and some other projects will update their documentation once I've got something to give them.

mavoga, 2025-04-25:
My opinion is that by forcing users to use an external script, they'll have to "reinvent the wheel" for the logic used to find the path of the sign-file tool (which is indeed the most recommended tool), etc. and risking to introduce vulnerabilities, when the most part has already been done - and largely tested - by you.

I really wonder what's preventing you from including a further sourceable variable to store the passphrase and exporting its content in the environment. After all, any solution for this issue will necessarily require injecting the KBUILD_SIGN_PIN variable into the environment, so why making everything so unnecessarily complicated?