dkms-project/dkms #364: forcing autoinstall to rebuild all modules

anbe42, 2023-11-24:
Writing with my Debian hat on ...

Whenever I upgrade a linux-headers-* package (and it does not change its name), I'd like to *rebuild* *all* dkms modules against these headers. Assuming that the ABI is stable if the ABI-version part in the package name (and kernel version) did not change is not a safe bet and the cases where the it actually causes problems are the nasty ones. (Also for kernels in experimental that always use an ABI version of 0, this assumption is outright wrong and modules *must* be rebuilt on upgrade.)

But dkms currently finds an old build of matching module/modversion/kvers/arch and keeps that. How can I best get around that? The relevant trigger would be that one activated from /etc/kernel/header_postinst.d/dkms.

Debian kernel maintainers are working on improving the situation, e.g. bumping the version in the package name much more often s.t. such required re-build events should be much less frequent (as there would be much more new-build events).

scaronni, 2023-11-29:
I think 99% of the people either use dkms in a very basic way (add, build, install and remove) of non-existing modules, or they go on the convoluted way of embedding scripts in the configuration files (for example https://github.com/openzfs/zfs/blob/master/scripts/dkms.mkconf and https://github.com/umlaeute/v4l2loopback/blob/main/dkms.conf).

I fall in the first category :)

Regarding the `original_module` part, that should be needed in case you need to replace running modules with newer versions from DKMS packages. The AMD provided drivers do that: https://www.amd.com/en/support/linux-drivers

anbe42, 2023-12-05:
I'm going to try this in `kernel_postinst.d_dkms.in`, right before calling `dkms_autoinstaller`:
```
case $0 in *header_postinst.d*)
    # unbuild all autoinstalled modules for this kernel to ensure they get
    # rebuilt against the updated headers by the next autoinstall below
    for mod_ver in $(dkms status -k "$inst_kern" 2>/dev/null | grep ': installed' | cut -d, -f1 | sort -u)
    do  
        dkms_conf="/var/lib/dkms/$mod_ver/source/dkms.conf"
        AUTOINSTALL=
        autoinstall=$(. "$dkms_conf" >/dev/null 2>&1; echo $AUTOINSTALL)
        test -n "$autoinstall" || continue
        dkms unbuild -k "$inst_kern" "$mod_ver"
    done
    ;;
esac
```
and then on Debian a rebuild of all modules for a single kernel can be triggered with `dpkg-reconfigure linux-headers-6.5.0-5-amd64`