dkms-project/dkms #392: Kernel match for BUILT_MODULE_NAME

dmipx, 2024-02-05:
We are building UVC kernel module.
To rebuild uvc modules prior to kernel 6.5 we needed only videodev and uvcvideo modules.
With kernel 6.5 there was refactoring of uvc framework and now we have another dependency - uvc module.
```
modinfo  /lib/modules/6.5.0-15-generic/kernel/drivers/media/usb/uvc/uvcvideo.ko | grep depends
depends:        videobuf2-v4l2,videodev,mc,uvc,videobuf2-common,videobuf2-vmalloc
```
Kernel 5.15, same as 6.2:
```
modinfo /lib/modules/5.15.0-25-generic/kernel/drivers/media/usb/uvc/uvcvideo.ko | grep depends
depends:        videobuf2-v4l2,videodev,videobuf2-common,videobuf2-vmalloc,mc
```

Getting this error for building on kernel 5.15.0 when having another module in conf file: `BUILT_MODULE_NAME[2]="uvc"`
`Error!  Build of uvc.ko failed for: 5.15.0-25-generic (x86_64)`

What is missing now is key DEPENDS, like **BUILT_MODULE_DEPENDS**:
```
BUILT_MODULE_NAME[1]="videodev"
BUILT_MODULE_LOCATION[1]="drivers/media/v4l2-core/"
DEST_MODULE_NAME[1]="videodev"
DEST_MODULE_LOCATION[1]="/kernel/drivers/media/v4l2-core/"

# Kernel 6.5
BUILT_MODULE_NAME[2]="uvc"
BUILT_MODULE_LOCATION[2]="drivers/media/common/"
BUILT_MODULE_DEPENDS[2]="^(6.[5])"
DEST_MODULE_NAME[2]="uvc"
DEST_MODULE_LOCATION[2]="/kernel/drivers/media/common/"
DEST_MODULE_DEPENDS[2]="^(6.[5])"
```

Thanks.

dmipx, 2024-02-05:
CLEAN[#] and CLEAN_MATCH[#] can be additional improvement.

dmipx, 2024-02-06:
Thanks for tips.
That means we need to maintain two DKSM packages - one for kernels below 6.5 and other for kernels above 6.5.
Our delivery is debian dpkg packaged DKMS, I cannot see such option to have both in one bundle..

anbe42, 2024-02-07:
One of the most sophisticated (and dynamically self-generating) dkms.conf I'v seen so far is
https://sources.debian.org/src/lttng-modules/2.13.11-1/debian/lttng-modules-dkms.dkms/?hl=1#L1

But for your problem its probably sufficient to do something conditional like
```
if <<<CONDITION>>> ; then
BUILT_MODULE_NAME[2]="uvc"
...
fi
```
(you have the complete bash syntax available for `<<<CONDITION>>>`)

dmipx, 2024-02-07:
Excellent!
I forgot that conf file is also bash syntax.

dmipx, 2024-02-08:
Resolved that with
```
# Kernel 6.5 introduced uvc module
if [ "$VERSION" -ge 6 -a "$PATCHLEVEL" -ge 5 ]; then
	BUILT_MODULE_NAME[2]="uvc"
	BUILT_MODULE_LOCATION[2]="drivers/media/common/"
	DEST_MODULE_NAME[2]="uvc"
	DEST_MODULE_LOCATION[2]="/kernel/drivers/media/common/"
fi
```

Issue can be closed.