dkms-project/dkms #409: OBSOLETE_BY does not prevent building

xypron, 2024-03-26:
The OBSOLETE_BY value is used in do_install() to prevent installing a module that is no longer needed or supported since the specified kernel version.

But the module is still being built.

It does not make much sense to build a module that will never be installed:

- The source code may not be compatible to the new kernel source due to changed kernel APIs.
- Building may take a considerable time.

We should not build modules for kernels which match the OBSOLETE_BY version.

timkb4cq, 2025-04-09:
At MX-Linux we package some realtek wifi dkms drivers, and by kernel 6.14 some of those now have good in-kernel drivers.  I was trying to use OBSOLETE_BY="6.14" in the dkms.conf files of those to avoid building unnecessary out of kernel modules but found out that OBSOLETE_BY doesn't stop the build - which fails because of the changes in 6.14.

If OBSOLETE_BY is not going to work why is it still in the documentation?  It certainly has obvious uses since specifying BUILD_EXCLUSIVE_KERNEL_MAX="6.13.9" throws an error which will cause a 6.14 kernel to not finish installing, which is, at a minimum, counterintuitive for such an option.

anbe42, 2025-04-09:
> specifying BUILD_EXCLUSIVE_KERNEL_MAX="6.13.9" throws an error

Which dkms version are you using and what is the error message you observe?
Can you provide a link to (the dkms.conf of) such a dkms module?

timkb4cq, 2025-04-10:
Dkms version 3.0.10-8+deb12u1 (debian bookworm)

dkms.conf
```
PACKAGE_NAME="rtl8821au"
PACKAGE_VERSION="5.12.5.2"
BUILT_MODULE_NAME[0]="8821au"
MAKE="kernelver=$kernelver ./dkms-make.sh"
CLEAN="'make' clean"
DEST_MODULE_LOCATION[0]="/updates/dkms"
AUTOINSTALL="yes"
BUILD_EXCLUSIVE_KERNEL_MAX="6.13.9"
```
Errors installing 6.14.1 liquorix kernel
```
Building module:
Cleaning build area...
kernelver=6.14.1-1-liquorix-amd64 ./dkms-make.sh.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.14.1-1-liquorix-amd64 (x86_64)
Consult /var/lib/dkms/8821au/5.12.5.2/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.14.1-1-liquorix-amd64 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.14.1-1-liquorix-amd64 (--configure):
 installed linux-image-6.14.1-1-liquorix-amd64 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 linux-headers-6.14.1-1-liquorix-amd64
 linux-image-6.14.1-1-liquorix-amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
A package failed to install.  Trying to recover:
Setting up linux-headers-6.14.1-1-liquorix-amd64 (6.14-2~mx23ahs) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 6.14.1-1-liquorix-amd64.
/usr/sbin/dkms.mx autoinstall --kernelver 6.14.1-1-liquorix-amd64
Sign command: /lib/modules/6.14.1-1-liquorix-amd64/build/scripts/sign-file
Binary /lib/modules/6.14.1-1-liquorix-amd64/build/scripts/sign-file not found, modules won't be signed

Building module:
Cleaning build area...
kernelver=6.14.1-1-liquorix-amd64 ./dkms-make.sh.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.14.1-1-liquorix-amd64 (x86_64)
Consult /var/lib/dkms/8821au/5.12.5.2/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.14.1-1-liquorix-amd64 failed!
run-parts: /etc/kernel/header_postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/header_postinst.d at /var/lib/dpkg/info/linux-headers-6.14.1-1-liquorix-amd64.postinst line 11.
dpkg: error processing package linux-headers-6.14.1-1-liquorix-amd64 (--configure):
 installed linux-headers-6.14.1-1-liquorix-amd64 package post-installation script subprocess returned error exit status 1
Setting up linux-image-6.14.1-1-liquorix-amd64 (6.14-2~mx23ahs) ...
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.14.1-1-liquorix-amd64.
/usr/sbin/dkms.mx autoinstall --kernelver 6.14.1-1-liquorix-amd64
Sign command: /lib/modules/6.14.1-1-liquorix-amd64/build/scripts/sign-file
Binary /lib/modules/6.14.1-1-liquorix-amd64/build/scripts/sign-file not found, modules won't be signed

Building module:
Cleaning build area...
kernelver=6.14.1-1-liquorix-amd64 ./dkms-make.sh.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.14.1-1-liquorix-amd64 (x86_64)
Consult /var/lib/dkms/8821au/5.12.5.2/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.14.1-1-liquorix-amd64 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.14.1-1-liquorix-amd64 (--configure):
 installed linux-image-6.14.1-1-liquorix-amd64 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 linux-headers-6.14.1-1-liquorix-amd64
 linux-image-6.14.1-1-liquorix-amd64

```

anbe42, 2025-04-10:
> Dkms version 3.0.10-8+deb12u1 (debian bookworm)

That version does not yet have BUILD_EXCLUSIVE_KERNEL_MIN/MAX, you need at least 3.0.11

anbe42, 2025-04-10:
You could use the old regex variant to express < 6.14.*:
`BUILD_EXCLUSIVE_KERNEL="^([1-5]\.|6\.([0-9]|1[0-3])\.)"`

anbe42, 2025-04-11:
Anyway, it sounds sensible to propagate OBSOLETE_BY to BUILD_EXCLUSIVE_KERNEL_MAX to prevent building the "unneeded" modules for "too new" kernels. Let's see if that approach breaks any users of OBSOLETE_BY.

commit 2ebf9ee8 (2025-04-11): dkms: propagate OBSOLETE_BY to BUILD_EXCLUSIVE_KERNEL_MAX

do not attempt to build modules that won't be installed

Fixes: #409
closed 2025-04-11 by scaronni