dkms-project/dkms #416: Failing to compile nvidia-open-dkms while using a ThinLTO/Clang kernel

ltsdw, 2024-05-15:
```
dkms: 3.0.12-1
clang: 17.0.6-2
```

As there been some debate over this on TKG and NVIDIA github (like https://github.com/NVIDIA/open-gpu-kernel-modules/issues/405), I'm facing something different, but I don't know whether it's related to dkms or not, for example, when trying to install nvidia-open-dkms on Arch Linux, I see this:

```
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
...
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
...
```

(It will keep printing a bunch of those, it will still successfully install the nvidia-open-dkms besides that)

Also trying to boot the system with the borked driver won't work (it will hang on boot).

So I tried replacing the `/usr/bin/strip` for `/usr/bin/llvm-strip`, and those assertions disappears and the nvidia-open kernel module installs normally, though it still borked (it will hang on boot)

I think maybe the dkms script isn't setting the right `STRIP=/usr/bin/llvm-stip` environment variable or other Clang related stuff?

danudey, 2024-05-17:
In order to successfully compile a kernel module via DKMS, you need to provide the module build with the same LLVM-related environment variables you used to build the kernel, and as far as I can tell this doesn't happen with DKMS (i.e. there's no way for DKMS to know what those parameters were).

On Ubuntu I had to modify the dkms file in `/usr/src/nvidia-<ver>/dkms.conf` to add those parameters; for example, to go from this:

```
MAKE[0]="'make' -j__JOBS NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES' KERNEL_UNAME=${kernelver} modules"
```

to this:

```
MAKE[0]="LLVM=1 'make' -j__JOBS NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES' KERNEL_UNAME=${kernelver} modules"
```

I would assume that on Arch you would have to do a similar thing.

(Ubuntu actually uses a different dkms.conf file which needs more work to fix.)

ltsdw, 2024-05-18:
This one works for me, although some of the envars may not be needed.

```patch
--- a/dkms	2024-05-18 11:52:29.624897615 -0300
+++ b/dkms	2024-05-18 11:52:17.387875553 -0300
@@ -589,17 +589,17 @@
     # Check if clang was used to compile or lld was used to link the kernel.
     if [[ -e $kernel_source_dir/vmlinux ]]; then
       if  readelf -p .comment $kernel_source_dir/vmlinux | grep -q clang; then
-        make_command="${make_command} CC=clang"
+        make_command="${make_command} CC=clang CXX=clang++ AR=llvm-ar OBJCOPY=llvm-objcopy"
       fi
       if  readelf -p .comment $kernel_source_dir/vmlinux | grep -q LLD; then
-        make_command="${make_command} LD=ld.lld"
+        make_command="${make_command} LD=ld.lld AR=llvm-ar OBJCOPY=llvm-objcopy"
       fi
     elif [[ -e "${kernel_config}" ]]; then
       if grep -q CONFIG_CC_IS_CLANG=y "${kernel_config}"; then
-        make_command="${make_command} CC=clang"
+        make_command="${make_command} CC=clang CXX=clang++ AR=llvm-ar OBJCOPY=llvm-objcopy"
       fi
       if grep -q CONFIG_LD_IS_LLD=y "${kernel_config}"; then
-        make_command="${make_command} LD=ld.lld"
+        make_command="${make_command} LD=ld.lld AR=llvm-ar OBJCOPY=llvm-objcopy"
       fi
     fi
 
@@ -1112,7 +1112,11 @@
         local built_module="$the_module$module_uncompressed_suffix"
         local compressed_module="$the_module$module_suffix"
 
-        [[ ${strip[$count]} != no ]] && strip -g "$built_module"
+        if [[ ${strip[$count]} != no ]] && [[ ${CC} == "clang" ]]; then
+            llvm-strip -g "$built_module"
+        elif [[ ${strip[$count]} != no ]]; then
+            strip -g "$built_module"
+        fi
 
         if (( do_signing )); then
             echo "Signing module $built_module"
```

PR #417 (2024-10-23): dkms: fix building nvidia open kernel modules against clang and thin/full lto compiled kernel
Right now there's some issues when trying to compile open-gpu-kernel-modules against a kernel compiled using Clang and Thin/Full LTO.

Also the script will pick the wrong version of the `strip` command otherwise the module will fail to compile successfully:
```
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion f [... cut]
commit 5aa01448 (2024-10-23): dkms: fix building against clang and thin/full lto
Right now there's some issues when trying to compile open-gpu-kernel-modules against a kernel compiled using clang+Thin/Full LTO.
Also the script will pick the wrong version of the strip command, is also necessary to set the OBJCOPY environment variable.

Fixes #416.
closed 2024-10-23 by scaronni