dkms-project/dkms #418: Leftover files after zfs build

SoongNoonien, 2024-05-31:
Hi!
I would have open a discussion but they seem to be disabled. I'm having troubles with this one: https://github.com/openzfs/zfs/issues/16221. Building the zfs kernel modules with dkms leaves some files in `/usr/lib/modules/KERNEL/build` behind. Do you have any idea what could be causing something like that?

xuzhen, 2024-06-01:
This is caused by automake's [automatic dependency tracking](https://www.gnu.org/software/automake/manual/automake.html#Dependencies).

You can disable it by adding `no-dependencies` into configure.ac
```
--- configure.ac	2024-06-01 06:49:59.000000000 +0000
+++ configure.ac	2024-06-01 06:49:59.000000000 +0000
@@ -40,7 +40,7 @@
 AC_CANONICAL_TARGET
 AM_MAINTAINER_MODE
 m4_ifdef([AM_SILENT_RULES], [AM_SILENT_RULES([yes])])
-AM_INIT_AUTOMAKE([subdir-objects foreign])
+AM_INIT_AUTOMAKE([subdir-objects foreign no-dependencies])
 # Remove default macros from config.h:
 #   PACKAGE, PACKAGE_{BUGREPORT,NAME,STRING,TARNAME,VERSION}, STDC_HEADERS, VERSION
 AC_CONFIG_HEADERS([zfs_config.h], [
```

closed 2024-06-01 by xuzhen

SoongNoonien, 2024-06-01:
Ok, thank you very much!