dkms-project/dkms #423: Keep the Module.symvers file along with the *.ko

DidierTrosset-Acqiris, 2024-06-20:
When the build is complete, only the `*.ko` files are copied to the final `module` directory.

However, some modules export functions, and to be able to use these exported functions when compiling other modules (either using dkms or not) one need the `Module.symvers` file.
This `Module.symvers` file is generated when compiling the module, but it is left in the `build` directory. Unfortunately, this directory is deleted when the build is finished.

I'd like to have an option such that the `Module.symvers` file is copied along the `*.ko` files to the final `module` directory.

DidierTrosset-Acqiris, 2024-07-23:
I am compiling my own module: a PCIe driver to a company made card. This card acquires data to internal memory and then transfers it to the host PC. The driver has been doing this successfully for more that 20 years.
I recently added the ability to transfer the data from our card directly to an nVidia GPU card using RDMA.

My precise use case is that my driver has to call functions that are exported by the nvidia driver to handle the direct data transfer from our company-card to the GPU-card.

My problem is that the compilation of the nvidia proprietary driver (that uses dkms) erases the `build` directory when done. It of course copies the build result (kernel module file) before deletion to a target directory. But the `Module.symvers` file resulting from the compilation of the nvidia driver is lost. That's the file the compilation of my own driver requires, and that I'd like to keep.

DidierTrosset-Acqiris, 2024-09-12:
I've not concatenated any symvers files.
What I did to make it work is compiling the nvidia driver manually (I mean not in the dkms automatic way, but in a shell) and checking a symvers was there. It's the standard nvidia driver in /usr/src/nvidia-current-525-147-05 installed by the nvidia-kernel-dkms Debian package.
Then I added "KBUILD_EXTRA_SYMBOLS=/usr/src/nvidia-current-525-147-05/Module.symvers" to the MAKE[0] make command of my module dkms.conf file.
Is that the part of interest?

What I can envision first is that dkms default behavior would be to copy the Module.symvers file alongside the final *.ko file. This need to be the default behavior of dkms. I mean it would not be practicable for me to change the nvidia dkms.conf file, adding a flag to conserve the symvers. I'd like the nvidia files installed by the apt package nvidia-kernel-dkms to be kept pristine.

Likewise, I'd want the final `/var/lib/dkms/nvidia-current/525.147.05/6.1.0-21-amd64/x86_64/module` directory to contain `Module.symvers` alongside the `*.ko` files.

That's the main thing I'd need, because as of now, I'm not using the Module.symvers that correspondsprecisely to the nvidia kernel module I'm using.

What I could envision next, is a way to indicate in the dkms.conf of my module, that it requires another dkms module, and that the KBUILD_EXTRA_SYMBOLS would be added automatically for me. Especially since the location of the final Module.symvers file can change. But I guess dkms has all the data it needs to create this string.

scaronni, 2024-11-26:
Can you provide a pull request with the changes you're interested in?

DidierTrosset-Acqiris, 2024-11-28:
Hi Simone. I created the pull request. Only three lines of code. https://github.com/dell/dkms/pull/459

closed 2024-11-30 by scaronni