dkms-project/dkms #426: In the loop, why is it a read-only variable?

wubo0067, 2024-07-03:
In the function make_tarball, it polls the elements in the array kernelver. Why are intree_module_dir and temp_module_dir read-only type variables?

```
make_tarball()
{
    ......
    for ((i=0; i<${#kernelver[@]}; i++)); do
        local intree_module_dir="$dkms_tree/$module/$module_version/${kernelver[$i]}/${arch[$i]}"
        local temp_module_dir="$temp_dir_name/dkms_main_tree/${kernelver[$i]}"
```

If it is a read-only variable, then using --all or specifying multiple -k kernel versions, only one kernel version will be packaged.

```
 ⚡ root@localhost  /usr/src  dkms mktarball -m i40e -v 2.25.9 --all
Marking modules for 4.18.0-425.3.1.el8.x86_64 (x86_64) for archiving...
_**/usr/sbin/dkms: line 1906: local: intree_module_dir: readonly variable
/usr/sbin/dkms: line 1907: local: temp_module_dir: readonly variable**_
Marking modules for 4.18.0-425.19.2.el8_7.x86_64 (x86_64) for archiving...

Marking /var/lib/dkms/i40e/2.25.9/source for archiving...
```

xuzhen, 2024-07-03:
There is already a PR for this: https://github.com/dell/dkms/pull/415

wubo0067, 2024-07-04:
The file name also needs to be optimized. If there are multiple -k or --all, it becomes too long, for example: i40e-2.25.9-kernel4.18.0-425.3.1.el8.x86_64-x86_64-kernel4.18.0-425.19.2.el8_7.x86_64-x86_64.dkms.tar

wubo0067, 2024-07-08:
1: Honestly, I'm not sure what kind of tests meet your requirements. If you have examples, I can give it a try.
2: Filename optimization refers to the situation where, under -k or -all, if there are multiple kernel patch versions in the system, the generated **xxx.dkms.tar** filename will be very long. I want to think about how to optimize it and try to make a PR.