dkms-project/dkms #427: nvidia-open-dkms: Strip assertion failed (ThinLTO compiled kernel) - Module succesfully compiled but can not boot

ptr1337, 2024-07-06:
Hi,

Somehow DKMS has problems stripping the kernel module, when the kernel is compiled with Thin or Full LTO.

DKMS is succesful at the compilation, but we are seeing warnings in the stripping part:
```
dkms install --no-depmod nvidia/555.52.04 -k 6.9.3-3-cachyos-lto
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:4131
strip: BFD (GNU Binutils) 2.42.0 assertion fail /usr/src/debug/binutils/binutils-gdb/bfd/elf.c:413
...
```

Compiling the Open Module directly at kernel compilation, like here:
https://github.com/CachyOS/linux-cachyos/blob/master/linux-cachyos/PKGBUILD#L550-L560

Works without issues, the module can be booted.

Booting the dkms compiled module does result into a not bootable kernel, e.g. graphical session does not work.

More Info:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/661

ltsdw, 2024-07-06:
It's a duplicate of #416 my PR #417 will fix it once it gets merged.

ptr1337, 2024-07-06:
Sorry, did not saw them. Thanks!

closed 2024-07-06 by ptr1337