dkms-project/dkms #450: invoke_command: dkms prints "terminated" message

Astartha, 2024-10-16:
latest dkms prints annoying "terminated" message, that comes from progress in invoke_
[dkms.patch](https://github.com/user-attachments/files/17401879/dkms.patch)
command.
example:

```
_Cleaning build area... done.                                                  
/usr/sbin/dkms: line 1132: 569570 Terminated              while true; do      
    sleep 3; echo -n ".";                                                     
done                                                                          
Building module(s).... done.                                                  
/usr/sbin/dkms: line 1132: 569615 Terminated              while true; do      
    sleep 3; echo -n ".";                                                     
done  (wd: /var/lib/dkms/snapapi26/0.8.43/build)                              
Signing module /var/lib/dkms/snapapi26/0.8.43/build/snapapi26.ko              
Signing module /var/lib/dkms/snapapi26/0.8.43/build/snumbd26.ko               
Cleaning build area... done.                                                  
/usr/sbin/dkms: line 1198: 570007 Terminated              while true; do      
    sleep 3; echo -n ".";                                                     
done  (wd: /var/lib/dkms/snapapi26/0.8.43/build)                              
                                                                              
Installing /lib/modules/5.15.0-206.153.7.1.el8uek.x86_64/extra/snapapi26.ko.xz
Installing /lib/modules/5.15.0-206.153.7.1.el8uek.x86_64/extra/snumbd26.ko.xz_
``` 

that comes from killed not being waited and that sends its output to stderr.

attached simple fix that waits for kill command. that works for me.

scaronni, 2024-10-21:
Interesting, it does not happen on any of my systems. Can you please make a merge request?

scaronni, 2024-10-21:
I've moved forward and made a pull request myself.