dkms-project/dkms #464: Module signing broken on upstream kernel 6.12 due to linux-headers .config removal

mihailim, 2024-12-11:
Upstream kernel commit https://github.com/torvalds/linux/commit/aaed5c7739be81ebdd6008aedc8befd98c88e67a removes the file names .config from generated header packages. Unfortunately, this breaks DKMS when secure boot is being used and any locally compiled the modules need to be signed.

DKMS sets its path to the kernel config into the kernel_config var in this function: https://github.com/dell/dkms/blob/93bf2bd79704c0a8a39b240a52b001610cdc419b/dkms.in#L274
DKMS then uses it to extract CONFIG_MODULE_SIG_HASH in this function: https://github.com/dell/dkms/blob/93bf2bd79704c0a8a39b240a52b001610cdc419b/dkms.in#L1021
If kernel_config isn't set, the build process issues "Kernel config ${kernel_config} not found, modules won't be signed" and skips signing the built .ko files, subsequently resulting in a runtime failure to load the corresponding module with "Loading of unsigned module is rejected" when secure boot is being used.

While it's possible to supply the path to the kernel config manually when invoking DKMS, it would need manual intervention after every kernel header package installation. This would be rather clumsy and inconvenient compared to the current set-and-forget approach where a kernel headers package install hook auto-builds and installs modules.

At Mr. Yamada's suggestion, I checked whether using `${kernel_source_dir}/include/config/auto.conf` works instead of `${kernel_source_dir}/.config`, and I can confirm it does -- build, signing, and installation proceed correctly.

Can we simply try to use `include/config/auto.conf` if it exists but `.config` does not?

Upstream does not wish to re-ship `.config` or to provide a symlink from `.config` to `include/config/auto.conf` to keep compatibility -- please see the last reply in the `linux-kbuild` mailing list thread discussing this issue here: https://lore.kernel.org/linux-kbuild/CAK7LNARqEOVOzP5vdUVF0KxQBNb9xtYs-COSXXWDMpBzGaLGow@mail.gmail.com/T/#m95f48caf46357a41c1df5e038e227a01ab89dbda

> The linux-headers package is intended to provide the necessary tools for building external modules.
> 
> The .config file is not required for building external modules, so it was removed.
> 
> This is an issue with DKMS.
> 
> Furthermore, I do not understand why they decided to allow the process to proceed successfully when the file DKMS requires is missing:
> 
> https://github.com/dell/dkms/blob/93bf2bd79704c0a8a39b240a52b001610cdc419b/dkms.in#L1026
> 
> They need to fix the issue in DKMS.

scaronni, 2024-12-12:
From which versions is `${kernel_source_dir}/include/config/auto.conf` available? Could you please prepare a merge request with the changes? Thanks.

xuzhen, 2024-12-12:
> From which versions is `${kernel_source_dir}/include/config/auto.conf` available? 

With https://github.com/torvalds/linux/commit/00c864f8903dd357471e8ee48f4d57aaa9a1a0de in 4.19, auto.conf will be guaranteed to exist.

scaronni, 2024-12-12:
Merge request done, please add comments there if you have any. The default is now the `auto.conf` file.

mihailim, 2024-12-12:
Whoa, this was lightning-quick! Thank you, @scaronni . The config file detection logic looks perfect, trying the newly-blessed-by-upstream-as-correct file before falling back on the old behavior. Can confirm that it works properly!

closed 2024-12-12 by scaronni