dkms-project/dkms #471: Conditional built modules in dkms.conf

JDarnley, 2024-12-19:
Similar to #392 I am looking for a conditional listing of files in dkms.conf but in contrast to that (I think) I want it to be based on a kernel config option.  In my case I specifically want CONFIG_AUXILIARY_BUS.

I make a package using "dkms mkdeb" so I need it to be checked when apt tries to make dkms install it much later and elsewhere for every kernel that dkms tried to install for.

Is there an approved way to query this at "build" time?

ian-abbott, 2025-08-11:
I'm not recommending the following as a recommended solution, but it is what I do in [Comedi](https://github.com/Linux-Comedi/comedi) as a workaround.

The `dkms.conf` file gets sourced in as Bash source after some sanity checks, so it is possible to put the `BUILT_MODULE_NAME[n]`, `BUILT_MODULE_LOCATION[n]`, and `DEST_MODULE_LOCATION[n]` entries inside a Bash `if/then/fi` command. One problem is that the array indices need to be in an increasing sequence without gaps, but one way to do that is to use a local variable to hold the current index, and increment it.

For example:

```
local nn
nn=0
BUILT_MODULE_NAME[$nn]="foo1"
BUILT_MODULE_LOCATION[$nn]="ziggy/"
DEST_MODULE_LOCATION[$((nn++))]="/kernel/ziggy/"
BUILT_MODULE_NAME[$nn]="foo2"
BUILT_MODULE_LOCATION[$nn]="ziggy/"
DEST_MODULE_LOCATION[$((nn++))]="/kernel/ziggy/"
if grep -q '^CONFIG_AUXILIARY_BUS=[ym]' ${kernel_source_dir}/.config' 2>/dev/null; then
BUILT_MODULE_NAME[$nn]="foo-aux"
BUILT_MODULE_LOCATION[$nn]="ziggy/"
DEST_MODULE_LOCATION[$((nn++))]="/kernel/ziggy/"
fi
```

scaronni, 2025-08-12:
@ian-abbott any chance you could test? You just need to convert your if line in a directive with the module name.

Thanks!

ian-abbott, 2025-08-13:
> [@ian-abbott](https://github.com/ian-abbott) any chance you could test? You just need to convert your if line in a directive with the module name.

Some of the tests I use in https://github.com/Linux-Comedi/comedi/blob/master/dkms.conf.in are not quite so straightforward, such as this one:

```
# Modules that require set_fs() support in the kernel
# (Try and mimic checks done by configure script.)
if test -d /lib/modules/${kernelver}/source ; then
	src="/lib/modules/${kernelver}/source"
else
	src="${kernel_source_dir}"
fi
if grep -q '^CONFIG_SET_FS=y' ${kernel_src_dir}/.config 2>/dev/null ||
	! test -f ${src}/include/asm-generic/uaccess.h ||
	! test -f ${src}/include/asm-generic/access_ok.h &&
	! grep -q CONFIG_SET_FS ${src}/include/asm-generic/uaccess.h 2>/dev/null
then
BUILT_MODULE_NAME[$nn]="serial2002"
BUILT_MODULE_LOCATION[$nn]="comedi/drivers/"
DEST_MODULE_LOCATION[$((nn++))]="/kernel/comedi/drivers/"
fi
```

scaronni, 2025-08-26:
https://github.com/dell/dkms/pull/533 implements this (will be in 3.3.0). Please reopen if necessary.

closed 2025-08-26 by scaronni