dkms-project/dkms #472: Failure to build a kernel module results in no dracut image being created on Fedora

belegdol, 2024-12-22:
Hello,

this has started happening recently on Fedora - looks like update to 3.1.3 being pushed is the first one exhibiting the incorrect behaviour. With the new version, failure to build xone results in the following:

```
>>> Scriptlet Ausgabe:
>>>
>>> Sign command: /lib/modules/6.12.5-200.fc41.x86_64/build/scripts/sign-file
>>> Signing key: /etc/pki/akmods/private/private_key.priv
>>> Public certificate (MOK): /etc/pki/akmods/certs/public_key.der
>>>
>>> Cleaning build area... done.
>>> Building module(s)...(bad exit status: 2)
>>> Failed command:
>>> make -j12 KERNELRELEASE=6.12.5-200.fc41.x86_64 -C /lib/modules/6.12.5-200.fc41.x86_64/build M=/var/lib/dkms/xone/v0.3-57-g29ec357/build
>>>
>>> Error! Bad return status for module build on kernel: 6.12.5-200.fc41.x86_64 (x86_64)
>>> Consult /var/lib/dkms/xone/v0.3-57-g29ec357/build/make.log for more information.
>>> Autoinstall on 6.12.5-200.fc41.x86_64 failed for module(s) xone(10).
>>>
>>> Error! One or more modules failed to install during autoinstall.
>>> Refer to previous errors for more information.
>>> /usr/lib/kernel/install.d/40-dkms.install failed with exit status 11.
>>>
>>> [RPM] %posttrans(kernel-core-6.12.5-200.fc41.x86_64) Scriptlet fehlgeschlagen, Beenden-Status 11
```

whereas with the working 3.0.13 the following message will get generated:

```
>>> Scriptlet Ausgabe:
>>> dkms: running auto installation service for kernel 6.12.6-200.fc41.x86_64
>>> Sign command: /lib/modules/6.12.6-200.fc41.x86_64/build/scripts/sign-file
>>> Signing key: /etc/pki/akmods/private/private_key.priv
>>> Public certificate (MOK): /etc/pki/akmods/certs/public_key.der
>>> 
>>> Building module:
>>> Cleaning build area...
>>> Building module(s)...(bad exit status: 2)
>>> Failed command:
>>> make -j12 KERNELRELEASE=6.12.6-200.fc41.x86_64 -C /lib/modules/6.12.6-200.fc41.x86_64/build M=/var/lib/dkms/xone/v0.3-57-g29ec357/build
>>> Error! Bad return status for module build on kernel: 6.12.6-200.fc41.x86_64 (x86_64)
>>> Consult /var/lib/dkms/xone/v0.3-57-g29ec357/build/make.log for more information.
>>> dkms autoinstall on 6.12.6-200.fc41.x86_64/x86_64 failed for xone(10)
>>> Error! One or more modules failed to install during autoinstall.
>>> Refer to previous errors for more information.
>>> dkms: autoinstall for kernel 6.12.6-200.fc41.x86_64 Failed. 
>>> dkms: running auto installation service for kernel 6.12.6-200.fc41.x86_64
>>> Sign command: /lib/modules/6.12.6-200.fc41.x86_64/build/scripts/sign-file
>>> Signing key: /etc/pki/akmods/private/private_key.priv
>>> Public certificate (MOK): /etc/pki/akmods/certs/public_key.der
>>> 
>>> Building module:
>>> Cleaning build area...
>>> Building module(s)...(bad exit status: 2)
>>> Failed command:
>>> make -j12 KERNELRELEASE=6.12.6-200.fc41.x86_64 -C /lib/modules/6.12.6-200.fc41.x86_64/build M=/var/lib/dkms/xone/v0.3-57-g29ec357/build
>>> Error! Bad return status for module build on kernel: 6.12.6-200.fc41.x86_64 (x86_64)
>>> Consult /var/lib/dkms/xone/v0.3-57-g29ec357/build/make.log for more information.
>>> dkms autoinstall on 6.12.6-200.fc41.x86_64/x86_64 failed for xone(10)
>>> Error! One or more modules failed to install during autoinstall.
>>> Refer to previous errors for more information.
>>> dkms: autoinstall for kernel 6.12.6-200.fc41.x86_64 Failed.
```

Failure to build an out-of-tree module is a regular occurrence and should not result in an unbootable grub entry being created as a result.

scaronni, 2024-12-25:
@belegdol can you test this simple fix on your system? Works on mine with a purposely broken module.

belegdol, 2024-12-26:
> @belegdol can you test this simple fix on your system? Works on mine with a purposely broken module.

Will do once 6.12.7 hits updates-testing.

belegdol, 2024-12-27:
I managed to test this earlier. With the patch, initramfs image was created properly. The output has changed slightly too:
```
>>> Scriptlet Ausgabe:
>>> Autoinstall of module xone/v0.3-57-g29ec357 for kernel 6.12.6-200.fc41.x86_64 (x86_64)
>>> 
>>> Sign command: /lib/modules/6.12.6-200.fc41.x86_64/build/scripts/sign-file
>>> Signing key: /etc/pki/akmods/private/private_key.priv
>>> Public certificate (MOK): /etc/pki/akmods/certs/public_key.der
>>> 
>>> Cleaning build area... done.
>>> Building module(s)...(bad exit status: 2)
>>> Failed command:
>>> make -j12 KERNELRELEASE=6.12.6-200.fc41.x86_64 -C /lib/modules/6.12.6-200.fc41.x86_64/build M=/var/lib/dkms/xone/v0.3-57-g29ec357/build
>>> 
>>> Error! Bad return status for module build on kernel: 6.12.6-200.fc41.x86_64 (x86_64)
>>> Consult /var/lib/dkms/xone/v0.3-57-g29ec357/build/make.log for more information.
>>> Autoinstall on 6.12.6-200.fc41.x86_64 failed for module(s) xone(10).
>>> 
>>> Error! One or more modules failed to install during autoinstall.
>>> Refer to previous errors for more information.
```

Thanks for the quick fix!

commit 2b91ef8f (2024-12-27): Fix #472, capture error output
closed 2024-12-27 by scaronni