dkms-project/dkms #483: Transfer DKMS repository to its own organization

scaronni, 2025-01-23:
Hi @gk4delltech you're listed as the only person for the Dell organization on Github: https://github.com/orgs/dell/people

We're trying to move DKMS to its own organization as it's no longer affiliated with Dell. It was abandoned with no people working on it and somehow through direct contact I managed to get access as an administrator and revive it.

As of now, it's been years since anyone from Dell had commit access to this repository.

I don't know who to contact as all the old Dell accounts are long gone from the repository access:

[image]

Can you help in trasnferring the repository in a new organization?

Thanks,
--Simone

anbe42, 2025-02-01:
+1

Once we get the repository moving, what do you think about renaming the `master` branch to `main` as well?

scaronni, 2025-02-03:
Sure,  no issue.

scaronni, 2025-03-28:
@anbe42 I've moved the `master` branch to `main` already.

gk4delltech, 2025-09-17:
hi @scaronni , I'm the open source governance for github.com/dell/    Somehow, I missed your January message.    I have escalated this internally.

gk4delltech, 2025-09-24:
@scaronni I've pinged the OSPO team again.

scaronni, 2025-09-30:
> Once we get the repository moving, what do you think about renaming the `master` branch to `main` as well?

@anbe42 et's do this and all the other changes you would like to see. Thanks.

closed 2025-09-30 by scaronni