dkms-project/dkms #484: Kernel not recongnized?

aleritty, 2025-01-23:
I'm having difficulties with DKMS, no module is getting built correctly and I get in ouput a strange message where the kernel version is simply not present:

```
# uname -a
Linux Cervello 6.12.4-1-MANJARO #1 SMP PREEMPT_DYNAMIC Mon, 09 Dec 2024 11:58:37 +0000 x86_64 GNU/Linux
#dkms --version 
dkms-3.1.3
```

```
# pacman -Q | grep headers
linux-api-headers 6.10-1
linux-ck-headers 6.11.10-1
linux612-headers 6.12.4-1
linux66-headers 6.6.65-1
opencl-headers 2:2024.10.24-1
vulkan-headers 1:1.4.303-1
```

```
# dkms autoinstall --all

Error! Your kernel headers for kernel  cannot be found at /usr/lib/modules//build or /usr/lib/modules//source.
Please install the linux-headers- package or use the --kernelsourcedir option to tell DKMS where it's located.

Error! Your kernel headers for kernel  cannot be found at /usr/lib/modules//build or /usr/lib/modules//source.
Please install the linux-headers- package or use the --kernelsourcedir option to tell DKMS where it's located.
Autoinstall on  failed for module(s) ryzen_smu(1) v4l2loopback(1).

Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
```

This presented suddenly without any system change, and now I'm without camera, bluetooth and for some reason after GDM handles the system to GNOME I'm left with a black screen (with X cursor) for 20 seconds.

Can you please help me debugging this issue?

anbe42, 2025-01-23:
> `# dkms autoinstall --all`

I don't think this has ever worked.
Either drop '--all' to build for the running kernel or replace it with explicit `-k <kernel_version>`.

aleritty, 2025-01-24:
Well it used to work for sure, I used it.
Anyway if I drop the --all I get nothing in output:
```
# dkms autoinstall --verbose
# uname -a
Linux Cervello 6.11.7-1-ck #1 SMP PREEMPT_DYNAMIC Sat Nov  9 12:30:33 EST 2024 x86_64 GNU/Linux
# dkms autoinstall -k 6.11.7-1-ck
#
```

And of course no modules are built.

anbe42, 2025-01-24:
What does a plain `dkms status` report?

aleritty, 2025-01-24:
I'm missing virtualbox modules, v4l2-loopback, and probably something else.
```
# dkms status
ryzen_smu/0.1.5, 6.11.7-1-ck, x86_64: installed
ryzen_smu/0.1.5, 6.12.4-1-MANJARO, x86_64: installed
ryzen_smu/0.1.5, 6.6.65-1-MANJARO, x86_64: installed
```

This is probably unrelated but at the same time this issue presented, my camera and bluetooth stopped working, I still have to test the realtek network card which used to need a module.

anbe42, 2025-01-24:
`dkms` does no longer seem to know about the other modules, otherwise `dkms status` should have listed them at least as `added`.
`ls -laR /var/lib/dkms` should confirm that.
Did you `dkms remove ...` something manually?

anbe42, 2025-01-24:
> Well it used to work for sure, I used it.

Even if `autoinstall --all` worked by chance in the past (I cannot reproduce that with a few old dkms versions I tried), it would never have worked for multiple kernels, simply because it only implements a loop over modules, not kernels. So I'm going to error out on invalid parameter combinations in the future.

closed 2025-01-28 by scaronni