dkms-project/dkms #488: /usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop

Nowa-Ammerlaan, 2025-02-05:
`autoinstall` outputs a bunch of warnings about improper use of `continue`:
```
nowa-gentoo-laptop nowa # dkms autoinstall -k 6.13.1-gentoo-dist
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
Sign command: /usr/src/linux/scripts/sign-file
Signing key: /root/kernel_key.pem
Public certificate (MOK): /root/kernel_key.pem

Autoinstall of module ni1045tr/24.8.0f118 for kernel 6.13.1-gentoo-dist (x86_64)

Error! Aborting build of module ni1045tr/24.8.0f118 for kernel 6.13.1-gentoo-dist (x86_64) due to missing BUILD_DEPENDS: nikal .
You may override by specifying --force.

...
```

It then refuses to build the kernel modules claiming a dependency is missing, yet according to `dkms status` this dependency was built successfully and is installed.

scaronni, 2025-02-05:
Ah, this is related to the change for getting the weak modules working again in SUSE (https://github.com/dell/dkms/blob/master/dkms.in#L1925). I wonder why I'm not getting it on any system I'm using this on (various iterations of Red Hat clones, Fedora, openSUSE / SLES, CBL Mariner / Azure Linux, Amazon Linux...).

Can you please do `bash -version`?

Nowa-Ammerlaan, 2025-02-05:
> Can you please do `bash -version`?

```
nowa@nowa-gentoo-laptop ~ % bash -version
GNU bash, version 5.2.37(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

scaronni, 2025-02-06:
Since I can't reproduce, can you make a quick test on your `/usr/sbin/dkms` by replacing line https://github.com/dell/dkms/blob/master/dkms.in#L1925 with:

```
if [[ $weak_modules_add && $weak_modules_remove ]]; then
```

And closing down the `if` block at https://github.com/dell/dkms/blob/master/dkms.in#L1960

Thanks.

scaronni, 2025-02-06:
Or test the patch at https://patch-diff.githubusercontent.com/raw/dell/dkms/pull/490.patch

Nowa-Ammerlaan, 2025-02-10:
Your PR resolves the bash problem, but the output of dkms looks weird now:
```
Autoinstall of module ni1045tr/25.0.0f175 for kernel 6.13.1-gentoo-dist (x86_64)
Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j8 KERNELRELEASE=6.13.1-gentoo-dist KERNELDIR=/lib/modules/6.13.1-gentoo-dist/build KERNELVER=6.13.1-gentoo-dist build

Error! Bad return status for module build on kernel: 6.13.1-gentoo-dist (x86_64)
Consult /var/lib/dkms/ni1045tr/25.0.0f175/build/make.log for more information.

Autoinstall on 6.13.1-gentoo-dist succeeded for module(s) ni488k ni843x nicmmk nicntdrk nicpciek nidimk nikal nimdbgk nimru2k nimxdfk nimxikniorbk nipalk nipxicidk nipxiethernet nipxifpk nipxigpk nipxim2 nipxirmk nismbus NiViPciK.
Autoinstall on 6.13.1-gentoo-dist failed for module(s) ni1045tr(10).
```

Before it would attempt to build and install all these modules, it failed (because of another issue I am still debugging). After the PR it stops after the first failure and then claims the remaining modules were built successfully.

scaronni, 2025-02-11:
Can you paste the `make.log` just after this happened?

anbe42, 2025-02-11:
please also paste `dkms status` output

Nowa-Ammerlaan, 2025-02-12:
> Can you paste the `make.log` just after this happened?

```
nowa-gentoo-laptop nowa # cat /var/lib/dkms/ni1045tr/25.0.0f175/build/make.log
DKMS (dkms-3.1.5) make.log for ni1045tr/25.0.0f175 for kernel 6.13.2-gentoo-dist (x86_64)
Wed 12 Feb 14:23:32 CET 2025
Cleaning build area
# command: make KERNELDIR=/lib/modules/6.13.2-gentoo-dist/build KERNELVER=6.13.2-gentoo-dist clean
Makefile:2143: !!!! WARNING !!!!
Makefile:2144: This kernel was configured and installed by the package manager.
Makefile:2145: "make" should not be run manually here.
Makefile:2146: See also: https://wiki.gentoo.org/wiki/Project:Distribution_Kernel
Makefile:2147: See also: https://wiki.gentoo.org/wiki/Kernel/Configuration
Makefile:2148: !!!! WARNING !!!!
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2143: !!!! WARNING !!!!
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2144: This kernel was configured and installed by the package manager.
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2145: "make" should not be run manually here.
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2146: See also: https://wiki.gentoo.org/wiki/Project:Distribution_Kernel
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2147: See also: https://wiki.gentoo.org/wiki/Kernel/Configuration
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2148: !!!! WARNING !!!!

# exit code: 0
# elapsed time: 00:00:00
Building module(s)
# command: make -j8 KERNELRELEASE=6.13.2-gentoo-dist KERNELDIR=/lib/modules/6.13.2-gentoo-dist/build KERNELVER=6.13.2-gentoo-dist build
Making ni1045tr.ko
Makefile:2143: !!!! WARNING !!!!
Makefile:2144: This kernel was configured and installed by the package manager.
Makefile:2145: "make" should not be run manually here.
Makefile:2146: See also: https://wiki.gentoo.org/wiki/Project:Distribution_Kernel
Makefile:2147: See also: https://wiki.gentoo.org/wiki/Kernel/Configuration
Makefile:2148: !!!! WARNING !!!!
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2143: !!!! WARNING !!!!
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2144: This kernel was configured and installed by the package manager.
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2145: "make" should not be run manually here.
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2146: See also: https://wiki.gentoo.org/wiki/Project:Distribution_Kernel
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2147: See also: https://wiki.gentoo.org/wiki/Kernel/Configuration
/usr/src/linux-6.13.2-gentoo-dist/Makefile:2148: !!!! WARNING !!!!
warning: the compiler differs from the one used to build the kernel
The kernel was built by: x86_64-pc-linux-gnu-gcc (Gentoo 14.2.1_p20241221 p6) 14.2.1 20241221
You are using:           gcc (Gentoo 14.2.1_p20241221 p6) 14.2.1 20241221
make[3]: *** No rule to make target 'ni1045tr-bin.o', needed by 'ni1045tr.o'.  Stop.
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-6.13.2-gentoo-dist/Makefile:1991: .] Error 2
make[1]: *** [Makefile:251: __sub-make] Error 2
make: *** [Makefile:76: ni1045tr.ko] Error 2

# exit code: 2
# elapsed time: 00:00:01
```

This failure is caused by some change in linux 6.13, it works for 6.12.

Nowa-Ammerlaan, 2025-02-12:
> please also paste `dkms status` output

```
nowa-gentoo-laptop nowa # dkms status
ni1045tr/25.0.0f175, 6.12.11-gentoo-dist, x86_64: installed
ni488k/25.0.0f95, 6.12.11-gentoo-dist, x86_64: installed
ni843x/25.0.0f98, 6.12.11-gentoo-dist, x86_64: installed
nicmmk/25.0.0f175, 6.12.11-gentoo-dist, x86_64: installed
nicntdrk/25.0.0f180, 6.12.11-gentoo-dist, x86_64: installed
nicpciek/25.0.0f174, 6.12.11-gentoo-dist, x86_64: installed
nidimk/25.0.0f151, 6.12.11-gentoo-dist, x86_64: installed
nikal/25.0.0f66, 6.12.11-gentoo-dist, x86_64: installed
nikal/25.0.0f66, 6.13.1-gentoo-dist, x86_64: installed
nikal/25.0.0f66, 6.13.2-gentoo-dist, x86_64: installed
nimdbgk/25.0.0f110, 6.12.11-gentoo-dist, x86_64: installed
nimru2k/25.0.0f151, 6.12.11-gentoo-dist, x86_64: installed
nimxdfk/25.0.0f110, 6.12.11-gentoo-dist, x86_64: installed
nimxik/25.0.0f178, 6.12.11-gentoo-dist, x86_64: installed
niorbk/25.0.0f110, 6.12.11-gentoo-dist, x86_64: installed
nipalk/25.0.0f108, 6.12.11-gentoo-dist, x86_64: installed
nipxicidk/25.0.0f180, 6.12.11-gentoo-dist, x86_64: installed
nipxiethernet/25.0.0f175, 6.12.11-gentoo-dist, x86_64: installed
nipxifpk/25.0.0f181, 6.12.11-gentoo-dist, x86_64: installed
nipxigpk/25.0.0f181, 6.12.11-gentoo-dist, x86_64: installed
nipxim2/25.0.0f175, 6.12.11-gentoo-dist, x86_64: installed
nipxirmk/25.0.0f181, 6.12.11-gentoo-dist, x86_64: installed
nismbus/25.0.0f180, 6.12.11-gentoo-dist, x86_64: installed
NiViPciK/25.0.0f125, 6.12.11-gentoo-dist, x86_64: installed
```

anbe42, 2025-02-12:
> make[3]: *** No rule to make target 'ni1045tr-bin.o', needed by 'ni1045tr.o'.  Stop.

> This failure is caused by some change in linux 6.13, it works for 6.12.

That sounds like an issue in the `ni1045tr/25.0.0f175` dkms module, not in dkms itself. Please report it to the module's upstream.

IIRC 6.13 had some changes regarding the directories used for the module build, maybe something now needs to have an absolute path.

Nowa-Ammerlaan, 2025-02-12:
> That sounds like an issue in the `ni1045tr/25.0.0f175` dkms module, not in dkms itself. Please report it to the module's upstream.

I know, my point was not that the module was failing to build, it did that before. The point is that it fails and stops immediately, yet the output of dkms claims it was a success after the https://github.com/dell/dkms/pull/490 was merged:

```
Autoinstall on 6.13.1-gentoo-dist succeeded for module(s) ni488k ni843x nicmmk nicntdrk nicpciek nidimk nikal nimdbgk nimru2k nimxdfk nimxikniorbk nipalk nipxicidk nipxiethernet nipxifpk nipxigpk nipxim2 nipxirmk nismbus NiViPciK.
Autoinstall on 6.13.1-gentoo-dist failed for module(s) ni1045tr(10).
```

To clarify, here's the full output before https://github.com/dell/dkms/pull/490:
```
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
/usr/bin/dkms: line 1910: continue: only meaningful in a `for', `while', or `until' loop
Sign command: /usr/src/linux/scripts/sign-file
Signing key: /root/kernel_key.pem
Public certificate (MOK): /root/kernel_key.pem

Autoinstall of module ni1045tr/25.0.0f175 for kernel 6.13.2-gentoo-dist (x86_64)
Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j8 KERNELRELEASE=6.13.2-gentoo-dist KERNELDIR=/lib/modules/6.13.2-gentoo-dist/build KERNELVER=6.13.2-gentoo-dist build

Error! Bad return status for module build on kernel: 6.13.2-gentoo-dist (x86_64)
Consult /var/lib/dkms/ni1045tr/25.0.0f175/build/make.log for more information.

Autoinstall of module ni843x/25.0.0f98 for kernel 6.13.2-gentoo-dist (x86_64)
Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j8 KERNELRELEASE=6.13.2-gentoo-dist KERNELDIR=/lib/modules/6.13.2-gentoo-dist/build KERNELVER=6.13.2-gentoo-dist build

Error! Bad return status for module build on kernel: 6.13.2-gentoo-dist (x86_64)
Consult /var/lib/dkms/ni843x/25.0.0f98/build/make.log for more information.

Autoinstall of module nicmmk/25.0.0f175 for kernel 6.13.2-gentoo-dist (x86_64)
Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j8 KERNELRELEASE=6.13.2-gentoo-dist KERNELDIR=/lib/modules/6.13.2-gentoo-dist/build KERNELVER=6.13.2-gentoo-dist build

Error! Bad return status for module build on kernel: 6.13.2-gentoo-dist (x86_64)
Consult /var/lib/dkms/nicmmk/25.0.0f175/build/make.log for more information.

Autoinstall of module nicntdrk/25.0.0f180 for kernel 6.13.2-gentoo-dist (x86_64)
Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j8 KERNELRELEASE=6.13.2-gentoo-dist KERNELDIR=/lib/modules/6.13.2-gentoo-dist/build KERNELVER=6.13.2-gentoo-dist build

Error! Bad return status for module build on kernel: 6.13.2-gentoo-dist (x86_64)
Consult /var/lib/dkms/nicntdrk/25.0.0f180/build/make.log for more information.
[... 76 lines cut ...]

Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
```

Versus after:
```
Sign command: /usr/src/linux/scripts/sign-file
Signing key: /root/kernel_key.pem
Public certificate (MOK): /root/kernel_key.pem

Autoinstall of module ni1045tr/25.0.0f175 for kernel 6.13.1-gentoo-dist (x86_64)
Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j8 KERNELRELEASE=6.13.1-gentoo-dist KERNELDIR=/lib/modules/6.13.1-gentoo-dist/build KERNELVER=6.13.1-gentoo-dist build

Error! Bad return status for module build on kernel: 6.13.1-gentoo-dist (x86_64)
Consult /var/lib/dkms/ni1045tr/25.0.0f175/build/make.log for more information.

Autoinstall on 6.13.1-gentoo-dist succeeded for module(s) ni488k ni843x nicmmk nicntdrk nicpciek nidimk nikal nimdbgk nimru2k nimxdfk nimxikniorbk nipalk nipxicidk nipxiethernet nipxifpk nipxigpk nipxim2 nipxirmk nismbus NiViPciK.
Autoinstall on 6.13.1-gentoo-dist failed for module(s) ni1045tr(10).
```

Note the difference in the list of modules that supposedly has succeeded or failed.

anbe42, 2025-02-12:
now I see your point ;-)

please try this:
in the `module_status_weak()` function in the dkms binary please add at the end `return 1` (around line 1972, between the closing `fi` and the final `}`

anbe42, 2025-02-12:
and if you go back to the version without #490, does the output of `dkms status` change?

scaronni, 2025-02-12:
> > please try this: in the `module_status_weak()` function in the dkms binary please add at the end `return 1` (around line 1972, between the closing `fi` and the final `}`
> 
> This resolves the problem on my end. The output of `dkms autoinstall` now looks the same again (minus the warnings about the incorrect usage of `continue`)

Just to be clear, did you just add `return 1` at the end of the function or also reverted the commit? Because you should not get the `continue` error anymore.

I've created https://github.com/dell/dkms/pull/493 to track attempts.

Nowa-Ammerlaan, 2025-02-12:
> Just to be clear, did you just add `return 1` at the end of the function or also reverted the commit? Because you should not get the `continue` error anymore.

Yes I only added `return 1`, this combined with your earlier fix made the `continue` warning go away while keeping the output otherwise the same.

scaronni, 2025-02-12:
Care to test latest git + https://patch-diff.githubusercontent.com/raw/dell/dkms/pull/493.patch? Thanks!

Nowa-Ammerlaan, 2025-02-13:
> Care to test latest git + https://patch-diff.githubusercontent.com/raw/dell/dkms/pull/493.patch? Thanks!

Works well on my end!

scaronni, 2025-02-13:
Awesome, thank you!

closed 2025-02-13 by scaronni