dkms-project/dkms #505: DKMS fails to insert module upon upgrading the kernel

norman-a, 2025-03-28:
I'm currently testing DKMS on a driver module, and it seems like DKMS is able to build and install the module whenever I install a different kernel version of ubuntu.

However, When I reboot and the new kernel is loaded, the module is not insertable:

```
~$ sudo modprobe enic
modprobe: ERROR: could not insert 'enic': Exec format error

~$ sudo dmesg | grep enic
[    3.462075] enic: disagrees about version of symbol module_layout
```

How can I resolve this?

Thanks everyone.

norman-a, 2025-03-28:
I figured out the solution in case anyone else runs into this issue. I would also like some confirmation on whether I'm doing it right.

In order to have the module be built against the new kernel that's being installed I added

``` KERNELDIR=/lib/modules/${kernelver}/build ```

to the end of my ```MAKE=``` line in dkms.conf.

kinghrothgar, 2025-03-28:
This is also the solution I came to: https://github.com/wisdpi/wp-ut5_dkms/pull/2/files

anbe42, 2025-03-29:
> I assume this is because DKMS still tries to build the module against the original kernel version and not the new one that's being installed.

There is a problem with your `dkms.conf` or `Makefile`, but without seeing your module source, I can't give advice.

anbe42, 2025-03-29:
> This is also the solution I came to: https://github.com/wisdpi/wp-ut5_dkms/pull/2/files. Interested to see if this is considered the standard approach.

This package is much more complicated that it would need to be. See e.g. the r8125-dkms Debian package as an example.

closed 2025-04-07 by norman-a