dkms-project/dkms #510: Regression in DKMS 3.1.7

christian-heusel, 2025-04-03:
All builds that I run with the latest DKMS version fail due to the following error:
```
$ sudo dkms install zfs/2.3.1 -k 6.12.21-1-lts
Sign command: /usr/lib/modules/6.12.21-1-lts/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Running the pre_build script....................... done.
Building module(s)....................... done.
Signing module /var/lib/dkms/zfs/2.3.1/build/module/zfs.ko
Signing module /var/lib/dkms/zfs/2.3.1/build/module/spl.ko
Running the post_build script...(bad exit status: 1)
Failed command:
cd /var/lib/dkms/zfs/2.3.1/build/ && /var/lib/dkms/zfs/2.3.1/build/scripts/dkms.postbuild -n zfs -v 2.3.1 -a x86_64 -k 6.12.21-1-lts -t /var/lib/dkms 
Consult /var/lib/dkms/zfs/2.3.1/build/make.log for more information.

$ cat /var/lib/dkms/zfs/2.3.1/build/make.log

Running the post_build script
# command: cd /var/lib/dkms/zfs/2.3.1/build/ && /var/lib/dkms/zfs/2.3.1/build/scripts/dkms.postbuild -n zfs -v 2.3.1 -a x86_64 -k 6.12.21-1-lts -t /var/lib/dkms 
cp: target '/var/lib/dkms/zfs/2.3.1/6.12.21-1-lts/x86_64/': No such file or directory

# exit code: 1
# elapsed time: 00:00:00
```

This error is new and was not present in the previous version:

```
# exit code: 0
# elapsed time: 00:01:01
Running the post_build script
# command: cd /var/lib/dkms/zfs/2.3.1/build/ && /var/lib/dkms/zfs/2.3.1/build/scripts/dkms.postbuild -n zfs -v 2.3.1 -a x86_64 -k 6.12.21-1-lts -t /var/lib/dkms 

# exit code: 0
# elapsed time: 00:00:00
Cleaning build area
# command: make -C module/ clean
make: Entering directory '/var/lib/dkms/zfs/2.3.1/build/module'
make -C /usr/lib/modules/6.12.21-1-lts/build M="$PWD"  clean
make[1]: Entering directory '/usr/lib/modules/6.12.21-1-lts/build'
  CLEAN   /var/lib/dkms/zfs/2.3.1/build/module/Module.symvers
make[1]: Leaving directory '/usr/lib/modules/6.12.21-1-lts/build'
rm -f Module.symvers Module.markers
find . -name '*.ur-safe' -type f -delete
make: Leaving directory '/var/lib/dkms/zfs/2.3.1/build/module'

```

anbe42, 2025-04-03:
What is dkms.postbuild good for?
It attempts to copy two files (Module.symvers is already copied by dkms nowadays), what is the header file zfs_config.h good for?

This probably broke because I now use a tempdir to populate the respective subtree s.t. we don't end up with something partial. 

scripts are only very rarely used by dkms packages and thus not well tested

christian-heusel, 2025-04-03:
This is the postbuild script for zfs: https://github.com/openzfs/zfs/blob/master/scripts/dkms.postbuild

anbe42, 2025-04-03:
I have a completely untested tentative fix

scaronni, 2025-04-04:
@christian-heusel can you test? @anbe42 feel free to release 3.1.7.1 or 3.1.8 with the fix.

christian-heusel, 2025-04-04:
The fix seems to work, I have rolled it out with the Arch Linux package and will report when there are any new issues 😊 👍🏻 

Thanks for coming up with a fix so fast, this is very much appreciated 🥰

anbe42, 2025-04-04:
I also confirmed the bug and fix on Debian.

Currently hunting another occurrence of "You cannot add the same module/version combo more than once." where remove finds nothing to remove but add chokes on something that is already there ...
That's not a new regression (hopefully it no longer happens that dkms goes into this state with my changes in 3.1.7), but there is more cleanup needed if dkms is already in that state.

jansohn, 2025-04-05:
> I also confirmed the bug and fix on Debian.
>
> Currently hunting another occurrence of "You cannot add the same module/version combo more than once." where remove finds nothing to remove but add [...]

is it related to https://github.com/dell/dkms/issues/384?

commit 3c9fb67f (2025-04-04): dkms: run post_build script after $mvka/module/ has been populated

The introduction of the intermediate temporary directory caused the
post_build script to be run at a time where $mvka/module/ was not yet
accessible by the script (as it was still below the tempdir).

This fixes the zfs-dkms regression.

Fixes: #510