dkms-project/dkms #512: module only installs for the current kernel

norman-a, 2025-04-07:
Hey guys,

I'm trying to have my module install on all the kernel versions of my host, however, it's only being installed on the current running one.

I've been using the postinst script from dkms-test-dkms package, and I see this line:

```
for DKMS_POSTINST in /usr/lib/dkms/common.postinst /usr/share/$DKMS_PACKAGE_NAME/postinst; do
        if [ -f $DKMS_POSTINST ]; then
                $DKMS_POSTINST $DKMS_NAME $DKMS_VERSION /usr/share/$DKMS_PACKAGE_NAME "" $2
                postinst_found=1
                break
        fi
done
```

Further inspecting ```/usr/lib/dkms/common.postinst```, I see the following:
```
KERNELS=$(ls -v /lib/modules/ 2>/dev/null || true)
CURRENT_KERNEL=$(uname -r)
```
```
if [ -z "$autoinstall_all_kernels" ]; then
    # If the current kernel is installed on the system or chroot
    if [ $(_is_kernel_name_correct $CURRENT_KERNEL) = "yes" ]; then
        if [ -n "$NEWEST_KERNEL" ] && [ ${CURRENT_KERNEL} != ${NEWEST_KERNEL} ]; then
            KERNELS="$CURRENT_KERNEL $NEWEST_KERNEL"
        else
            KERNELS=$CURRENT_KERNEL
        fi
    # The current kernel is not useful as it's not installed
    else
        echo "It is likely that $CURRENT_KERNEL belongs to a chroot's host"

        # Let's use only the newest kernel if this is not a first installation
        # otherwise build for all kernels
        if [ -n "$NEWEST_KERNEL" -a -n "$UPGRADE" ]; then
            KERNELS="$NEWEST_KERNEL"
        fi
    fi
fi
```

I'm assuming this variable ```$autoinstall_all_kernels``` being empty has to do with the fact that dkms isn't installing my module for all the kernels, but I'm not sure where I could set it. Also manually running dkms build and install without specifying the kernel version does the same.

Where can I configure dkms so that my modules get installed for all kernels?

anbe42, 2025-04-07:
/etc/dkms/framework.conf, override with config snippets in /etc/dkms/framework.conf.d/

norman-a, 2025-04-07:
@anbe42 so this is what I did:
I wrote a file custom.conf, with "autoinstall_all_kernels=yes" and saved it under /etc/dkms/framework.conf.d/

This still doesn't generate the change in frameworke.conf. What am I supposed to do to trigger it?

Also, instead of manually doing this in /etc/dkms/, could i have my postinst script do it?

anbe42, 2025-04-08:
/etc/dkms/framework.conf.d/*.conf is read by dkms after /etc/dkms/framework.conf, it does not need or cause framework.conf to be regenerated. For common.postinst to read framework.conf.d/*.conf, too, you need dkms 3.1.2 or later (otherwise put your changes directly in framework.conf)

Framework settings are in the scope of the distribution and/or the local admin, a module package should not take influence there. (For Debian and derivatives, I'm now (upcoming Debian 13 (trixie)) enabling autoinstall_all_kernels by default in the dkms package to avoid confusion. But I'm reluctant to do that for all distributions by default, especially if they support multiple module versions to be installed concurrently)

norman-a, 2025-04-08:
@anbe42 thank you makes sense.

closed 2025-04-08 by norman-a