dkms-project/dkms #521: Is dkms being skipped the correct behaviour if kernel headers aren't installed?

norman-a, 2025-04-18:
For example, if I run ```$apt install linux-image-6.8.0-45-generic``` I get the following:

```
dkms: autoinstall for kernel 6.8.0-45-generic was skipped since the kernel headers for this kernel do not seem to be installed
```

Installing both the image and headers ( ```$apt install linux-image-6.8.0-45-generic linux-headers-6.8.0-45-generic``` ) successfully triggers dkms module autoinstallation. 

What's the reasoning behind this? Is there a way to activate dkms autoinstallation without directly installing kernel headers along with the kernel image?

Thanks in advance for the responsiveness I really appreciate it.

scaronni, 2025-04-18:
No, your modules need to be compiled against the kernel headers, so there is no way around that. With a packaged DKMS, these dependencies are installed automatically.

closed 2025-04-18 by scaronni

anbe42, 2025-04-21:
There are usually some (unversioned) meta-packages available in the distribution that help you keep image and headers up-to-date and in sync. Most distributions provide several kernel flavors, so you need to pick the headers package matching your image package - that cannot be modeled by a dependency in the dkms package.

E.g. in Debian amd64 there are `linux-image-amd64` and `linux-headers-amd64`, `linux-image-cloud-amd64` and `linux-headers-cloud-amd64`, `linux-image-rt-amd64` and `linux-headers-rt-amd64`.